micropython/micropython #1979: Move to next version, v1.7?

dpgeorge, 2016-04-09:
It's been over 2 months since the last version, v1.6 (31st Jan 2016) with 388 commits, so time for the next one!

There have been a lot of changes, most notably to the esp8266 port.  I suggest going straight to v1.7 for this.

Any objections?  Anything to merge before tagging v1.7?

pfalcon, 2016-04-10:
+1 for 1.7

inaugurator, 2016-04-10:
Sounds good!

dpgeorge, 2016-04-11:
v1.7 will be tagged imminently.  No pushing until it's done!

dpgeorge, 2016-04-11:
Last minute tests show PYBv1.0 failing on basics/int_big_{div,mul}.py.  Need to investigate....

dpgeorge, 2016-04-11:
Failing int_big_div/mul are very difficult to reproduce.  Seems to be related to a USB serial problem because there are seven 0's that are not printed in one place but are printed in another.  Will probably defer fix to a later date.

pfalcon, 2016-04-11:
So, is fix for nt_big_div/mul is deferred or 1.7 deferred? If it can't be tagged immediately, I propose to postpone it until esp8266 port release.

dpgeorge, 2016-04-11:
v1.7 has been tagged at 9b0a150bd69f9c62a14045b4a2ecfaaf7523f542.

closed 2016-04-11 by dpgeorge