micropython/micropython #1982: Heads up, packaged in Fedora

jhdulaney, 2016-04-13:
Just as a heads up, I've packaged micropython in Fedora.

pfalcon, 2016-04-13:
@jhdulaney : Great news! Can you please elaborate if you packaged it to your private repository, or into official Fedora repos (so everyone can install it with "yum" right away?). Also, probably worth posting it on a forum, where more people will see the info: http://forum.micropython.org/viewforum.php?f=2 . Thanks!

jhdulaney, 2016-04-13:
It's in the main repositories for Fedora 24 (which is currently Alpha) and Rawhide (what will be F25+).

All you need to do is:
sudo dnf -y install micropython

dpgeorge, 2016-04-13:
Awesome!

pfalcon, 2016-04-28:
Posted as http://forum.micropython.org/viewtopic.php?f=8&t=1834 .

closed 2016-04-28 by pfalcon