micropython/micropython #1983: mpy-tool.py is not Python 2.x compliant

peterhinch, 2016-04-14:
A build using persistent bytecode fails with
```
CC build-PYBV10/pins_PYBV10.c
MPY scripts/fonts.py
Creating build-PYBV10/scripts/frozen_mpy.c
  File "../tools/mpy-tool.py", line 234
    print('   ', end='')
                    ^
SyntaxError: invalid syntax
make: *** [build-PYBV10/scripts/frozen_mpy.c] Error 1
make: *** Deleting file `build-PYBV10/scripts/frozen_mpy.c'
[adminpete@axolotl]: /mnt/qnap2/data/Projects/MicroPython/micropython/stmhal
```
This can be fixed by mandating Python3 in py/mkenv.mk but then ``make deploy`` fails unless I reinstate Python2. This is because (as I understand it) the ``usb.core`` module supports only P2.

SpotlightKid, 2016-04-14:
This particular problem can be fixed by inserting `from __future__ import print_function` at the top of the file before any other statement.

dpgeorge, 2016-04-14:
mpy-tool.py already forces Python3 (see https://github.com/micropython/micropython/blob/master/tools/mpy-tool.py#L1) so the variable in py/mkenv.mk should not take any effect here.  So there's something else wrong with your set-up.

> This is because (as I understand it) the usb.core module supports only P2.

usb is supported for Python3.  I use it without issue.

peterhinch, 2016-04-15:
Odd, my system honours that shebang if I run a Python script from Bash.

I don't really claim to understand the makefile, but it does appear to be specifying the Python interpreter from that imported by mkenv.mk:
```
$(BUILD)/$(FROZEN_MPY_DIR)/frozen_mpy.c: $(addprefix $(BUILD)/,$(FROZEN_MPY_FILES:.py=.mpy))
	@$(ECHO) "Creating $@"
	$(Q)$(PYTHON) $(MPY_TOOL) -f -q $(BUILD)/genhdr/qstrdefs.preprocessed.h $^ > $@
```

As for usb.core I had difficulty locating a source for this module. A web search led me [here](https://github.com/walac/pyusb). Has this been superseded?

dpgeorge, 2016-04-15:
> I don't really claim to understand the makefile, but it does appear to be specifying the Python interpreter from that imported by mkenv.mk:

Yes you are correct!  I mis-read the Makefile.  Will try to fix it.

As for usb, I just use `pip3 install pyusb` (as root).

dpgeorge, 2016-04-15:
Python 2.7 should now work with mpy-tool, see c3beb16db33e941021ff839a2fa9c806c134aa48.

peterhinch, 2016-04-15:
It works fine :)

closed 2016-04-15 by peterhinch