micropython/micropython #1984: make install fails on Mac OS X

muellermartin, 2016-04-15:
Compiling MicroPython v1.7 works fine on Mac OS X, but running the `install` target of the `Makefile` in the subfolder `unix` fails with following error (relevant excerpt):

    install -D micropython /usr/local/Cellar/micropython/1.7/bin/micropython
    install: illegal option -- D
    usage: install [-bCcpSsv] [-B suffix] [-f flags] [-g group] [-m mode]
                   [-o owner] file1 file2
           install [-bCcpSsv] [-B suffix] [-f flags] [-g group] [-m mode]
                   [-o owner] file1 ... fileN directory
           install -d [-v] [-g group] [-m mode] [-o owner] directory ...
    make: *** [install] Error 64

This is likely due to the incompatible flag `-D` which is only available for the `install` command contained in GNU coreutils, but not in the BSD version of `install`.

As solution I suggest to run `install -d` on the install directory first and then call `install` for the two files seperately. I've successfully used this approach for the package in the Homebrew package manager (see: Homebrew/homebrew-core#353).

pfalcon, 2016-04-28:
Fixed by linked pullreq.

closed 2016-04-28 by pfalcon

hardkrash, 2020-04-08:
This problem is occurring again with the master branch.

The -D option is not portable.