micropython/micropython #1986: Current _boot.py not compatible with ESP8266-01

doudz, 2016-04-17:
Current _boot.py is not compatible with ESP8266-01, probably because there's no filesystem on it ?

ioctl(5, 0)
readblocks(0, 3ffeeb00(4096))
readblocks(0, 3ffee850(4096))
Traceback (most recent call last):
  File "", line 9, in <module>
  File "", line 13, in check_bootsec
  File "", line 14, in readblocks
OSError: 5
ioctl(1, 0)
ioctl(5, 0)
readblocks(0, 3ffef050(4096))
NLR jump failed

dpgeorge, 2016-04-17:
The problem is that the `esp.flash_read()` call fails when checking for an existing filesystem (that's the OSError: 5).  The system then continues as if the FS was mounted and tries to execute `boot.py` from the filesystem (which didn't mount properly).  This leads to another exception but this is not caught and the system hangs.

pfalcon, 2016-04-17:
@doudz : Filesystem on modules with 512K flash is not supported (assuming yours is such). To help us diagnose it, move away _boot.py, build, boot, run "import port_diag", and provide its output.

christian-peace, 2016-04-17:
Hello,
@pfalcon : Thank you, i had test your descripton above:
----------------------------------------------------------------------------
>>> import port_diag
Flash ID: 1640e0 (Vendor: e0 Device: 4016)
Flash bootloader data:
Byte @2: 00
Byte @3: 00 (Flash size: 512KB Flash freq: 40MHZ)
----------------------------------------------------------------------------
My Board "LoLin Node Mcu" has also only 512k. I'll buy another one with 4m.

Christian

doudz, 2016-04-17:
I got this :
MicroPython v1.7-90-g8007f84-dirty on 2016-04-17; ESP module with ESP8266
Type "help()" for more information.
>>> import port_diag
Flash ID: 1440e0 (Vendor: e0 Device: 4014)
Flash bootloader data:
Byte @2: 00
Byte @3: 00 (Flash size: 512KB Flash freq: 40MHZ)

dpgeorge, 2016-04-17:
@christian-peace your flash id is the same as my Adafruit Huzzah feather board, and the "16" at the start indicates that it's 2**0x16=4mbyte in size.

@doudz your flash id indicates that it's 2**0x14=1mbyte in size.

BUT, both of your devices are configured to use only 512kb of the flash, that's why the filesystem is failing.  You need to change the internal configuration so it can use all of the flash.

dpgeorge, 2016-04-17:
I just tried with one of my ESP boards and it's possible to modify the configuration byte so that the board can use all the flash:
```
import esp
buf = bytearray(4096) # allocate buffer for 1 sector of flash
esp.flash_read(0, buf) # read sector 0
buf[3] = (buf[3] & 0xf) | (x << 4) # REPLACE x with: 2 for 1mb, 3 for 2mb, 4 for 4mb
esp.flash_erase(0) # erase sector 0
esp.flash_write(0, buf) # write sector 0
```
You can then do a hard reset and `import port_diag` to check it worked.

Note that if you mess up your config you can restart fresh by erasing all flash (esptool.py erase_flash) and re-deploying the uPy firmware.

pfalcon, 2016-04-17:
> You need to change the internal configuration so it can use all of the flash.

To expand on what @dpgeorge says, this configuration should be set properly (match actual flash size used) at the time of firmware flashing. `make deploy`, the official way to flash firmware after the build, sets suitable parameters automatically. If you use other means to flash, you need to set flashing parameters accordingly.

pfalcon, 2016-04-17:
> it's possible to modify the configuration byte so that the board can use all the flash:

For our (maintainers') sanity, I'd refrain from such advices ;-). We don't what happens next, maybe next few milliseconds vendor SDK functions reach for that byte, use it to calculate system param storage address, find there nothing, crash (or overwrite something with something, such that crashes happen later which is only worse).

Vendor says that those params should be set during flashing, vendor uses those params to calculate important data. Unless we can change that calculation (we can't), we just follow the official process and tell everyone else to do the same.

doudz, 2016-04-18:
Oh, I did't realize that `make deploy` is doing something different that me.
Since I use a virtual machine to build micropython (https://github.com/adafruit/esp8266-micropython-vagrant), I used the following command to flash:
`python esptool.py -p /dev/ttyUSB0 write_flash 0x00 firmware-combined.bin`

But the correct one seems to be 
`python esptool.py -p /dev/ttyUSB0 --baud 115200 write_flash --flash_size=8m 0 firmware-combined.bin`

dpgeorge, 2016-04-18:
> Vendor says that those params should be set during flashing, vendor uses those params to calculate important data.

It seems that the byte to configure the flash size is not read again after booting.  It requires a hard reset to reconfigure the SDK to use this new info.

dpgeorge, 2016-04-18:
> For our (maintainers') sanity, I'd refrain from such advices ;-).

I think it's good to let people know what's going on underneath the hood.  Giving users more information can't hurt :)

christian-peace, 2016-04-18:
Thanks for the information. It worked according to the disclosure `--flash_size=32m`.
I'am using the virtual machine like @doudz 
```
>>> import port_diag
Flash ID: 1640e0 (Vendor: e0 Device: 4016)
Flash bootloader data:
Byte @2: 00
Byte @3: 40 (Flash size: 4MB Flash freq: 40MHZ)
```
It was my fault, I'm sorry.

Regards Christian

doudz, 2016-04-18:
it works good for me too
>>> import port_diag
Flash ID: 1440e0 (Vendor: e0 Device: 4014)
Flash bootloader data:
Byte @2: 00
Byte @3: 20 (Flash size: 1MB Flash freq: 40MHZ)

So definitely  not a micropython issue

closed 2016-04-19 by doudz