micropython/micropython #1988: Error after reset

christian-peace, 2016-04-17:
After compiling (no errors/warnings), firmware flash and reset, I received the following error message:
d��|��;snc��$��b�dc䌜���$��c�lsdl$��|��soc��d$l���b�c�ܜ��d��c��dsd�d��|��{rnc��l���#�c�Ĝ��㌄c�dc$sd��sl���bl�b;|�cld���lb��o��g��l�|�l�$��$������l`�'����c��b���cl�c�l{ls�n�����g����c{#��b���c�x��l{l��d�ioctl(1, 0)
ioctl(5, 0)
readblocks(0, 3ffeeb00(4096))
readblocks(0, 3ffee850(4096))
Traceback (most recent call last):
  File "", line 9, in <module>
  File "", line 13, in check_bootsec
  File "", line 14, in readblocks
OSError: 5
ioctl(1, 0)
ioctl(5, 0)
readblocks(0, 3ffef050(4096))
NLR jump failed

What I have overlooked or done wrong?

This is a great project . Thank you very much.

peterhinch, 2016-04-17:
Github issues are intended for specific bug reports. For support you're better off asking in the [forum](http://forum.micropython.org/index.php). I'd spell out what hardware you're using and how you're connecting to it.

christian-peace, 2016-04-17:
Hello Peter,
Excuse me, I forgot to specify the hardware and the connection.
I'm using an ESP8266 (MCULua V3 Board) connected via usb.
The connection to the REPL is not possible, after reset the message appears.

Thanks for your help.

Adixylian, 2016-04-17:
same here
Huzzah esp8266 breakout

dpgeorge, 2016-04-17:
Yes, this is the same as #1986.  Closing this one because the other was first :)

closed 2016-04-17 by dpgeorge