micropython/micropython #1989: frozen bytecode generation fails

robert-hh, 2016-04-17:
Just a small thing, maybe not worth fixing:
if the path given for FROZEN_MPY_CODE contains .., e.g- "../scripts", the generated frozen_mpy.c will not compile. The auto-created symbols will contain _.._ 
Otherwise, a very useful feature. Thanks a lot for that.

dpgeorge, 2016-04-18:
Yes, FROZEN_MPY_CODE must be a directory within stmhal.  Patches to Makefile to fix this are welcome!

robert-hh, 2016-04-18:
I agree that this issue notification is more to be seen a a help for someone running into that prblem. But changing line 473 of mpy_tool.py into:
`        rc.freeze(rc.source_file.str.replace('.', 'd').replace('/', '_')[:-3] + '_')
`removed the compile error, by replacing '.' by 'd' (or some other valid character). On the downside the created scripts directory would not be in the build folder, so Makefile will have to be changed too.

dpgeorge, 2016-11-07:
@robert-hh I actually made a patch yesterday for this.

dpgeorge, 2016-11-07:
By that, I mean I have the patch locally, not yet pushed :)

robert-hh, 2016-11-07:
Trying to support you on cleaning the backlog, I closed it, considering it as of low importance.

dpgeorge, 2016-11-08:
> Trying to support you on cleaning the backlog, I closed it, considering it as of low importance.

Thank you, that's much appreciated.  But we should keep tickets for things that can be improved open (such as this one), otherwise someone else will run into the problem (like I did the other day) and open a new ticket and the discussion starts again.