micropython/micropython #1992: unix: "function does not take keyword arguments" message can mislead

peterhinch, 2016-04-18:
If function kw arguments are supplied from a dict, and the dict (erroneously) contains unexpected arguments, the above error message is rather misleading. 
```python
d = {'argument' : 'A', 'bar' : 0}

def foo(*, bar=False):
    print('foo')
foo(**d)
```
Cpython produces the self-evident
```
>>> foo(**d)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: foo() got an unexpected keyword argument 'argument'
>>> 
```

peterhinch, 2016-04-19:
I've encountered another manifestation of this. Its appearance depends on following the method of input exactly, with only the function def being entered in paste mode. It also requires the arg name to be long.
```
[adminpete@axolotl]: /mnt/qnap2/data/Projects/MicroPython/micropython/unix
$ ./micropython
MicroPython v1.7-64-gb122ed0-dirty on 2016-04-18; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> bar = {'averylongnamehere': 11, 'g': 77}
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== def foo(*,averylongnamehere = 1, b = 2, c = 3, d = 4, e = 5, f = 6, g = 7):
===     print(averylongnamehere,b,c,d,e,f,g)
=== 
=== 
>>> foo(b = 222, **bar)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function does not take keyword arguments
>>>
```
In practise this error crops up periodically. As dictionary expansion is typically used for functions having many args locating the cause is tedious. I have found it necessary to use short names in real code. And it would be very helpful, where an unexpected argument is in the dict, for the error message to name the arg.

dpgeorge, 2016-04-20:
Your second piece of code is a real bug and is different from the misleading nature of the error message.  See #1998.

dpgeorge, 2017-03-22:
Original issue fixed by 58f23def55a705357b352ce8af642e03cd5c49f0

closed 2017-03-22 by dpgeorge