micropython/micropython #1995: esp8266: tx/rx pins not reset on board reset

deshipu, 2016-04-19:
I experimented a little bit with using the rx and tx (gpio3 and gpio1) pins for other purposes than UART, and I noticed that when the board resets (due to a crash or soft reset), the pins are not initialized back to act as UART again.

dpgeorge, 2016-05-13:
The only error I can see is that the RX pin is not reinitialised on soft reset.  Everything else seems to work.

To test that TX is reset correctly:
```
>>> import machine
>>> machine.Pin(1, machine.Pin.IN)
```
REPL then no longer prints stuff, but doing ctrl-D does work, does a soft reset, and TX pin is working again.

To test that RX is reset on hard-reset:
```
>>> import machine
>>> machine.Pin(3, machine.Pin.IN)
Pin(3)
>>> 
```
REPL could print the result of creating the pin, but then becomes unresponsive.  A hard reset fixes it.

To test that RX is not reset on soft-reset:
```
>>> def f():
...     import machine
...     machine.Pin(3, machine.Pin.IN)
...     raise SystemExit
```
REPL comes up after soft-reset but you can't type anything.

It seems to be this line that's the issue: https://github.com/micropython/micropython/blob/master/esp8266/uart.c#L56 .  It probably should initialise the RXD pin instead of RTS.

pfalcon, 2016-05-28:
@dpgeorge : Ack, looks like that.

dpgeorge, 2016-05-29:
Should be fixed.  Tested using the description in the post above.

closed 2016-05-29 by dpgeorge