micropython/micropython #1996: esp8266: crash when using gpio16 in i2c

deshipu, 2016-04-19:
The board crashes when gpio16 is passed as either sda or scl to the ``machine.I2C`` initialization:

```
>>> from machine import I2C, Pin
>>> i2c = I2C(sda=Pin(16), scl=Pin(5))
Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x40253da2, epc2=0x00000000, epc3=0x00000000, excvaddr=0xffffffff, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 29616, room 16 
tail 0
chksum 0xd6
load 0x3ffe8000, len 980, room 8 
tail 12
chksum 0xaa
ho 0 tail 12 room 4
load 0x3ffe83e0, len 360, room 12 
tail 12
chksum 0x70
csum 0x70
```

deshipu, 2016-04-19:
This is going to repeat in a lot of places, probably -- I wonder if it would make sense to have some kind of generic "GPIO16NotSupported" exception.

dpgeorge, 2016-04-20:
The alternative is to support gpio16.  I would assume that most people use gpio16 to wake from deep sleep, but maybe that's not the case?

deshipu, 2016-04-20:
I think some nodemcu boards have a LED on gpio16. It certainly would be better to support using that pin -- not all use cases need the deep sleep, after all, and the number of pins is not that great, so every additional one counts. I have no idea how costly that would be, though -- as I understand, that pin is handled differently than the other pins, so you would need special cases everywhere?

One thing I'm sure is it should either work or throw an error, and not just quietly ignore it or crash...

dpgeorge, 2016-04-22:
GPIO16 is now supported for I2C.  See 67a6d31955cb7ff8985f9f96675a70870eea4089.

closed 2016-04-22 by dpgeorge

tintin10q, 2022-07-06:
I have the same issue coming with this code 

```python
from machine import I2C, Pin
i2c = machine.I2C(freq=9600,scl=Pin(5), sda=Pin(4))
i2c.scan()
```

I have no idea what to do there is nothing connected to pins 4 and 5.

peterhinch, 2022-07-07:
I2C requires pullup resistors. The value is not critical - typically something around 4.7KΩ to 3.3V.