micropython/micropython #2000: REPL suggests 'Type "help()"...' even if no help implemented

infinnovation, 2016-04-20:
The pyexec REPL implementation prints a welcome message on startup whose second line is
```
Type “help()” for more information.
```
However, it appears that some ports (esp8266, “minimal” and pic16bit) do not implement a help function.  This could result in a poor experience for the first-time user.

Before trying to address this, there are several issues to consider:

- In the case where no help function is available, pxexec could either not print the second line at all, or print something generic e.g. “See micropython.org/docs for more information.”
- The test for a help function could be at compile-time or run-time.  A run-time test would allow a port to implement help() in python via a startup file, at the expense of some extra code.
- In the case of a compile-time test, the mapping of the help function could be within a #if clause in modbuiltins.py rather than via MICROPY_PORT_BUILTINS in mpconfigport.h

Any thoughts?

toolmacher, 2016-04-21:
Why not include a help function to esp8266, minimal pic16bit with a small text, which can be extended when more functions supported.

infinnovation, 2016-04-21:
I had in mind that some ports may purposely not implement help in order to reduce code size as much as possible.  Maybe I am over-thinking it.

So on another tack I have submitted PR #2002 to assist in implementing help for other ports.

toolmacher, 2016-04-22:
or perhaps a dummy function of help in main.c, with no output of any text or object description.

example: 
STATIC const char *help_text = "";
STATIC mp_obj_t common_help(uint n_args, const mp_obj_t *args){
if(n_args == 0){ printf("%s",help_text); }
return mp_const_none;}
MP_DEFINE_CONST_FUN_OBJ_VAR_BETWEEN(mp_builtin_common_obj,0,1, common_help);

And rename pyb_help to common_help.

infinnovation, 2016-04-22:
As written, that's no help at all!  I'd at lease put a small help text such as "See micropython.org\n"

pfalcon, 2017-04-16:
This was fixed by 211244d1f3.

closed 2017-04-16 by pfalcon