micropython/micropython #2004: UART.readline() returns buffer if \n not in buffer

ryannathans, 2016-04-22:
Expected behavior: return None if \n not in buffer

eg:

```
while True:
    print(pyb.UART(2).readline()) #outputs one character at a time (if UART set to non-blocking)

```

peterhinch, 2016-04-22:
In the case of the Pyboard this behaviour is by design, is documented and depends on the timeout value. In the case of buffered reading, if the timeout elapses without a newline occurring it returns the characters buffered up to that time.

If None were returned and flow control were in operation, the user would need to take steps to prevent the buffer from filling and stalling the transmission. This would involve checking for data (with any()) and emptying the buffer to restart transmission. The caller would still need to rebuild the line from received fragments.

Having readline() behave as it does makes this responsibility explicit.

Comments requested.

ryannathans, 2016-04-22:
@peterhinch that makes makes a lot of sense for blocking reads (non zero timeout).

I don't feel as though this is correct for non blocking reads (timeout=0). It seems to wait until the next character is received before returning.

peterhinch, 2016-04-22:
It behaves as documented here. With pins X1 and X2 linked (Pyboard Lite UART 2 txd and rxd) I get an immediate return:
```
/mnt/qnap2/data/Projects/MicroPython/projects/tft> repl
Entering REPL. Use Control-X to exit.
>
MicroPython v1.6-292-gf6ec01c-dirty on 2016-04-13; PYBLITEv1.0 with STM32F411RE
Type "help()" for more information.
>>> 
>>> from pyb import UART
>>> uart = UART(2, 9600, timeout =0)
>>> uart.write('abc')
3
>>> uart.any()
3
>>> uart.readline()
b'abc'
>>> 
```
A subsequent uart.readline() also returns immediately (with None). I can't replicate a situation where timeout=0 results in an indefinite wait.

ryannathans, 2016-04-22:
Will test further in the morning... something fishy going on here on my side.

ryannathans, 2016-04-29:
Bug in my code regarding buffer size. Whoops

closed 2016-04-29 by ryannathans