micropython/micropython #2005: esp8266 won't stop trying to connect to given AP with a wrong password

jumasheff, 2016-04-22:
When I was initializing wlan connection I have mistyped my wi-fi router's password and esp8266 kept trying to connect to the network even after rebooting the board. I had to reflash the firmware to shut it up :)

Build and board info:

Al-Cloud Inside (black) board.

MicroPython v1.7-145-g3bc9b57 on 2016-04-23; ESP module with ESP8266
>>> import port_diag
Flash ID: 1440e0 (Vendor: e0 Device: 4014)
Flash bootloader data:
Byte @2: 00
Byte @3: 20 (Flash size: 1MB Flash freq: 40MHZ)

>>> os.uname()
(sysname='ESP8266', nodename='ESP8266', release='1.5.2(80914727)', version='v1.7-145-g3bc9b57 on 2016-04-23', machine='ESP module with ESP8266')

>>> esp.meminfo()
data  : 0x3ffe8000 ~ 0x3ffe83d4, len: 980
rodata: 0x3ffe83e0 ~ 0x3ffe8548, len: 360
bss   : 0x3ffe8548 ~ 0x3fff4eb8, len: 51568
heap  : 0x3fff4eb8 ~ 0x3fffc000, len: 29000

dpgeorge, 2016-04-22:
This is the way the underlying wifi sub-system works: it will keep on trying to connect to the AP that was last configured, even if it was wrongly configured (eg wrong password).

To turn off the error messages use:
```
import esp
esp.osdebug(None)
```

pfalcon, 2016-05-28:
As explained above, this is the behavior of the vendor SDK. The way to turn off debug logging is provided, and next KS backer release will come with debugging off by default.

closed 2016-05-28 by pfalcon