micropython/micropython #2010: OneWire (DS18B20) library is not working

mkarliner, 2016-04-25:
The example code:

import time
ds = onewire.DS18B20(ow)
roms = ds.scan()
ds.start_measure()
time.sleep_ms(750)
for rom in roms:
    print(ds.get_temp(rom))

Consistantly gives a value of 23.0625

Using the same hardare, board and sensor and the Arduino environment with this library https://github.com/PaulStoffregen/OneWire works as expected.

mkarliner, 2016-04-25:
My apologies for a rubbish first issue!
Hardware nodemcu 0.9 (ESP12), DS18B20 with pullup on vero.
Software v04
Driver as built in (??)
At this point I'm assuming that this page http://docs.micropython.org/en/latest/esp8266/esp8266/quickref.html#onewire-driver
documents the valid state of the software. If there is somewhere else I should be
looking to know what is functional in the current release, please do say.

cheers

Mike

dpgeorge, 2016-04-25:
Ok, thanks for the info.  Which GPIO pin did you use?  What value pull-up resistor?  Did you power the DS18B20 via VCC or use parasitic power from the data pin?  Yes, the driver is built in if you use alpha-v04.

mkarliner, 2016-04-25:
OK, mea culpa, my bad, etc, etc.

I'd typed in the example code at the repl, not thinking that the timing is critical.
When def'd as a function and  called, it works

<blush>

dpgeorge, 2016-04-25:
Great!

closed 2016-04-25 by dpgeorge

dpgeorge, 2016-04-28:
> I'd typed in the example code at the repl, not thinking that the timing is critical.
> When def'd as a function and called, it works

The actual problem is not that it doesn't work at the REPL (it does) it's just that you need to call the `convert_temp()` method (formerly `start_measure()`) each time you want a new temperature reading.