micropython/micropython #2011: Intent to publish mbed port

infinnovation, 2016-04-25:
As I mentioned in a [forum topic](http://forum.micropython.org/viewtopic.php?f=12&t=1781), I'm working on a port of MicroPython to the mbed platform.

Since I'm getting close to being able to publish this on developer.mbed.org, I thought it best to check with the devs that I'm not going to upset anyone by doing this, whether there are any particular issues I need to be aware of, etc.

The port is in github on the ["mbed" branch of infinnovation/micropython](https://github.com/infinnovation/micropython/tree/mbed).

There are several more commits I need to make, but this should give you an idea of what I'm about.
In the README-mbed.md file I've tried to make it clear that this is not (yet?!) an official MicroPython port and is somewhat rough around the edges.

Any feedback is welcome.

dpgeorge, 2016-04-25:
I see that you are targeting the classic mbed HAL, not the more recent mbed OS.  I don't know for how long the classic mbed will be supported (probably a while), but I guess there are plenty of boards out there that would be enabled by this port.

The main comment I have is that you should try to implement the classes in machine module, not just wrap the mbed class names.  For example Serial should be UART and DigitalOut should be Pin.  That way all MicroPython ports have the same libraries and work in (roughly) the same way.

infinnovation, 2016-04-25:
I'm targetting the mbed classic platform because that is where my FRDM-K64F took me when I plugged it in for the first time.  I would have dearly loved to have MicroPython available to help explore the board's capabilities.  I'm certainly aware of the newer mbed OS but haven't had time to get to grips with it or its yotta build system yet.

From the mbed standpoint, it makes sense (to me at least) to have consistent naming across languages and toolsets.  The idea being that you could prototype an application in MicroPython and then fairly easily rewrite it in C++ if required.  But from the MicroPython standpoint, it clearly makes sense to have consistent naming across ports and boards.  So I will look at implementing both - just needs some more of that scarce resource: time.

infinnovation, 2016-04-28:
FYI I published my mbed port on developer.mbed.org last night and posted an announcement in their forum, but the latter hasn't been approved by the moderator yet.

kfricke, 2016-04-28:
Try our forum then! 😎

dpgeorge, 2016-05-13:
mbed port is published.  See forum link above for further discussion.

closed 2016-05-13 by dpgeorge