micropython/micropython #2016: esp8266 - ESP-01 has problems with the onewire library

stucchimax, 2016-04-27:
I have an ESP-01 where I loaded the onewire.py script from the repository.  The firmware is compiled with a checkout from yesterday and today, and I'm having some issues:

First off, when I try to import onewire, I get this:
```python
>>> from machine import Pin
>>> import onewire
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "onewire", line 147, in <module>
  File "onewire", line 134, in test
  File "onewire", line 52, in scan
MemoryError: memory allocation failed, allocating 636 bytes
>>> import onewire
>>>
```
As you can see, the second time it actually works.

Then, when I try to run a scan, it always reports memory errors, as follows:

```python
>>> import time
>>> ow = onewire.OneWire(Pin(0))
>>> ds = onewire.DS18B20(ow)
>>> ds.scan()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "onewire", line 98, in scan
  File "onewire", line 52, in scan
MemoryError: memory allocation failed, allocating 696 bytes
>>> ds.scan()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "onewire", line 98, in scan
  File "onewire", line 52, in scan
MemoryError: memory allocation failed, allocating 856 bytes
>>> ds.scan()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "onewire", line 98, in scan
  File "onewire", line 52, in scan
MemoryError: memory allocation failed, allocating 596 bytes
```

It's strange to see that the quantity of memory always changes, even from board to board.  I tried this on different boards, and they all show the same behaviour.

```python
>>> esp.freemem()
26992
```
This is the output of freemem.  Hope this can help.

In case, I'm available for more tests/info.

dpgeorge, 2016-04-27:
Which onewire driver did you use?  You should try using this one: esp8266/tests/onewire.py.  Copy it to the esp8266/scripts/ directory, build, and reflash.

stucchimax, 2016-04-27:
That's exactly what I did.

Is there anything you would like me to do to help in debugging, then ?

dpgeorge, 2016-04-27:
Ok, thanks, I see the problem now.  The issue is that it runs the onewire test when you import the script, and because you don't have anything connected to Pin(12) the onewire.scan() function thinks that 256 devices are connected and tries to make a big list of all their ROM addresses (and runs out of memory doing it).

In 8ed3a9eb9c02149ace9616813306fd3940ae032c I stopped the test running on import, and it should now work!

stucchimax, 2016-04-27:
Great!  I started a build now, and will try tomorrow and will report here.

Thanks!

dpgeorge, 2016-04-28:
The tests/onewire.py script has been moved to the scripts/ directory, and the driver itself has been modified a bit.  Please try with this new code.  Refer to the quickref at http://docs.micropython.org/en/latest/esp8266/esp8266/quickref.html to get the 1-wire driver working.

dpgeorge, 2016-05-03:
Should be fixed.

closed 2016-05-03 by dpgeorge