micropython/micropython #2017: generator.throw() raises SIGSEGV

slush0, 2016-04-27:
Hello,

high level motivation of this example code is to have only one instance of "func1" (which is for simplicity just doing `yield True`) running at the time. There's `layout` decorator which take care of that - it calls `throw()` on previously run generator to kindly ask for stop himself.

`_current_layout.throw(Exception)` throws `StopIteration` exception with the result of first-started generator, as expected. But then `layout` decorator of `gen1` function have to finish. Once it touch `ret` variable, code crashes, because it probably points to invalid reference - result of the generator was already given as a result of `throw()`.

Pardon me if my analysis is wrong, because I don't understand the underlying C code, but it seems there's some issue with handling results of generators stopped by `throw` method.

Following code crashes on SIGSEGV:

```python
_current_layout = None

def layout(func):
    def inner(*args, **kwargs):
        global _current_layout

        print("Launching layout")
        if _current_layout:
            print("Layout %s exists, stopping it" % _current_layout)
            try:
                _current_layout.throw(Exception)
            except Exception as e:
                print(type(e))
                print("Stopped existing layout: %s" % str(e))

        gen = func(*args, **kwargs)
        _current_layout = gen

        try:
            ret = yield from gen
        except Exception as e:
            print("Exception %s" % str(e))

        # It crashes on SIGSEGV here, after touching 'ret'
        print("Layout %s asked to run layout %s" % (gen, ret))
        return ret
    return inner

@layout
def func1():
    while True:
        print("inner func")
        try:
            yield True
        except Exception:
            print("Stopping...")
            break

    print("...Stopped")
    return("Result of func1")

def run():
    gen1 = func1()
    print(gen1.send(None))

    gen2 = func1()
    print(gen2.send(None))
    print(gen1.send(None))

run()
```

pfalcon, 2016-04-27:
Quick comment from top of my head: in Python, you cannot "recursively" activate already active (i.e. running) generator. Well, it's not really "in Python", that operation just doesn't make sense for coroutines in general. If you try to do something like, CPython throws exception, but uPy doesn't have all the bloated checks, and may crash in that case. So, if you're doing something like that above (I can't tell ;-) ), don't.

slush0, 2016-04-27:
Well, I think this code should work in general (I'm not aware of "activating already activated generator). In the end, we're probably going to solve it in another, less "asynchronous" way, but this example still uncovers some problematic behavior, so I would appreciate if you take a look at it.

pfalcon, 2016-04-27:
Well, the simplest testcase is:

~~~
def gen():
    return 1
    yield

f = gen()

def run():
    print((yield from f))
    print((yield from f))

next(run())
~~~

And that's it - yield from doesn't like already terminated generators.

slush0, 2016-04-27:
Is there any clean way how to know if the generator is already terminated?

pfalcon, 2016-04-27:
Fixed in eff85bb1dcca0602cef055986ea3b8192dea1d71. Thanks for the report.

closed 2016-04-27 by pfalcon

slush0, 2016-04-27:
Thank you, that was amazing speed :-O