micropython/micropython #2018: micropython build failing: xtensa-lx106-elf-gcc: error: unrecognized command line option '-mforce-l32'

Narcolapser, 2016-04-28:
I have two ESP8266-12 modules. I decided I wanted to use micropython with them so I mucked around and got micropython to build and flashed module 1. I fiddled around with it for the rest of the day and was pleased. went to bed. The next day I come back to this project and was pondering if I should flash module 2 now. I cd into micropython/esp8266 and type make and I got the mentioned error:
`xtensa-lx106-elf-gcc: error: unrecognized command line option '-mforce-l32'`

I found this patch to the open-sdk: https://github.com/pfalcon/esp-open-sdk/blob/master/1000-mforce-l32.patch
But it refused to apply:
```
toben@oxygen:/opt/Espressif/crosstool-NG$ git apply 6b0c9f92fb8e11c6be098febb4f502f6af37cd35.patch
error: gcc/config/xtensa/constraints.md: No such file or directory
error: gcc/config/xtensa/xtensa.c: No such file or directory
error: gcc/config/xtensa/xtensa.md: No such file or directory
error: gcc/config/xtensa/xtensa.opt: No such file or directory
```

I"m posting this to micropython because micropython is what isn't compiling, crosstool-NG compiled fine. micropython seems to be calling a feature that doesn't exist? If this is actually a crosstool-NG issue I'll take it there. I'm just not even sure where to start. I'm just so confused how I got it to work once and now it's broken! help. :(

dpgeorge, 2016-04-28:
It should work automagically: try recloning the esp-open-sdk repo and rebuilding it (just type `make`).

pfalcon, 2016-04-28:
Yes, just use the current esp-open-sdk. See also: http://forum.micropython.org/viewtopic.php?f=16&t=1689

closed 2016-04-28 by pfalcon

Narcolapser, 2016-04-29:
That's weird. I downloaded and built esp-open-sdk three days ago and it said nothing was new today. But rebuilding it worked so it's all good I guess.

PriankaAg, 2016-09-27:
cp: cannot remove `crosstool-NG/local-patches/gcc/4.8.5/1000-mforce-l32.patch': Permission denied
make: *** [/home/Desktop/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 1

dhylands, 2016-09-27:
You probably ran something previously using sudo and now you're trying to run without, and the patch file is owned by root.

PriankaAg, 2016-09-27:
how to fix this error .....
should i do permission to normal user if so ... then how and what command i use...??

dhylands, 2016-09-27:
```
sudo chown OWNER filename
```
replace OWNER with the username that you're using. You can use:
```
sudo chown -R OWNER dirname
```
to recursively change ownership of all files in a directory tree.