micropython/micropython #2020: usart init of port3 stm not correct

rolandvs, 2016-04-29:
As mentioned in the link at the forum, mpy does not initialize the USART3 port correctly. When using the alternative function of the GPIO for an UART, you have the possibility to "move around" the pins. One option is to have the RXD/TXD on the GPIOC and the RTS/CTS lines on GPIOB. The init code doesn't support this feature.

I would opt to add the `MICROPY_HW_UART3_HSK` to `mpconfigboard.h` and make the changes as included in `uart.c`.

```
// RXD/TXD ok, handshake lines cannot be configured
#define MICROPY_HW_UART3_PORT (GPIOC)
#define MICROPY_HW_UART3_PINS (GPIO_PIN_10 | GPIO_PIN_11) // PORTC
// separate GPIO for handshake pins
#define MICROPY_HW_UART3_HSK (GPIOB)
#define MICROPY_HW_UART3_RTS  (GPIO_PIN_14)   // PORT B
#define MICROPY_HW_UART3_CTS  (GPIO_PIN_13)   // PORT B

```
Fragment from micropython/stmhal/uart.c
```
#if defined(USART3) && defined(MICROPY_HW_UART3_PORT) && defined(MICROPY_HW_UART3_PINS)
        // USART3 is on PB10/PB11 (CK,CTS,RTS on PB12,PB13,PB14), PC10/PC11 (CK on PC12), PD8/PD9 (CK on PD10)
        case PYB_UART_3:
            UARTx = USART3;
            irqn = USART3_IRQn;
            GPIO_AF_UARTx = GPIO_AF7_USART3;
            GPIO_Port = MICROPY_HW_UART3_PORT;
            GPIO_Pin = MICROPY_HW_UART3_PINS;
            #if defined(MICROPY_HW_UART3_RTS)
            if (uart_obj->uart.Init.HwFlowCtl & UART_HWCONTROL_RTS) {
       
                //##GPIO_Pin |= MICROPY_HW_UART3_RTS;
                // we need to set the GPIO pins on a different GPIO than the UART
                GPIO_Pin2 = MICROPY_HW_UART3_RTS;
                GPIO_Port2 = MICROPY_HW_UART3_HSK;
                
            }
            #endif
            #if defined(MICROPY_HW_UART3_CTS)
            if (uart_obj->uart.Init.HwFlowCtl & UART_HWCONTROL_CTS) {

                //##GPIO_Pin |= MICROPY_HW_UART3_CTS;
                // we need to set the GPIO pins on a different GPIO than the UART
                GPIO_Pin2 = MICROPY_HW_UART3_CTS;
                GPIO_Port2 = MICROPY_HW_UART3_HSK;

            }
            #endif
            __USART3_CLK_ENABLE();
            break;
        #endif
```

As mentioned in the forum:     http://forum.micropython.org/viewtopic.php?f=2&t=1826

dpgeorge, 2016-05-06:
On pyboard PC10/PC11 are used for the SD card, so that's why there's no need to support these alternative pins for the UART tx/rx.

So the issue is that you want to be able to configure it for your own, custom board, right?

How about just using the Pin object and reconfiguring it using the alt argument to init (or the constructor)?

rolandvs, 2016-11-25:
@dpgeorge I'm sorry I forgot about this. But to answer the question:

I think having a  ```mpconfigboard.h``` that defines the proper connection of the uarts and other things is what we need and not have these function implemented using a Pin object in a later stage. The fact of the matter is that the UART handshake pins are already defined in the ```mpconfigboard.h```, but do not cover all permitted combinations. By splitting the ```CTS/RTS``` gpio assignment from the UART ```RXD/TXD``` pins all possible cases are handled correctly. Of course you are right that the SD card on the PYB board already uses the pins, but any other board probably not. I have included the fragment again also due to the fact the code above has an error :-)

```
/* fragment from mpconfigboard.h */
// RXD/TXD ok, handshake lines cannot be configured
#define MICROPY_HW_UART3_PORT (GPIOC)
#define MICROPY_HW_UART3_PINS (GPIO_PIN_10 | GPIO_PIN_11) // PORTC
// separate GPIO for handshake pins
#define MICROPY_HW_UART3_HSK (GPIOB)
#define MICROPY_HW_UART3_RTS  (GPIO_PIN_14)   // PORT B
#define MICROPY_HW_UART3_CTS  (GPIO_PIN_13)   // PORT B

/* fragment from stmhal/uart.c */
#if defined(USART3) && defined(MICROPY_HW_UART3_PORT) && defined(MICROPY_HW_UART3_PINS)
        // USART3 is on PB10/PB11 (CK,CTS,RTS on PB12,PB13,PB14), PC10/PC11 (CK on PC12),
        // PD8/PD9 (CK on PD10), but CTS/RTS can be on other PORT (GPIOB)
        case PYB_UART_3:
            UARTx = USART3;
            irqn = USART3_IRQn;
            GPIO_AF_UARTx = GPIO_AF7_USART3;
            GPIO_Port = MICROPY_HW_UART3_PORT;
            GPIO_Pin = MICROPY_HW_UART3_PINS;
            #if defined(MICROPY_HW_UART3_RTS)
            if (uart_obj->uart.Init.HwFlowCtl & UART_HWCONTROL_RTS) {
                GPIO_Pin2 |= MICROPY_HW_UART3_RTS;
                GPIO_Port2 = MICROPY_HW_UART3_HSK;
            }
            #endif
            #if defined(MICROPY_HW_UART3_CTS)
            if (uart_obj->uart.Init.HwFlowCtl & UART_HWCONTROL_CTS) {
                GPIO_Pin2 |= MICROPY_HW_UART3_CTS;
                GPIO_Port2 = MICROPY_HW_UART3_HSK;
            }
            #endif
            __USART3_CLK_ENABLE();
            break;
        #endif
```

dhylands, 2016-11-25:
Hey Roland,

We've been slowly moving all of the peripheral drivers to use pin objects rather the port/pin designators.
While I was reworking the teensy port, I switched it over the way @dpgeorge was referring to. You can see that it now looks much cleaner in the mpconfigboard.h file:
https://github.com/dhylands/micropython/blob/teensy-usb/teensy/boards/TEENSY_3.6/mpconfigboard.h#L30-L34

By using pin objects, it means you get the port and pin mask combined into one entity. You can see this is used in the teensy uart.c here:
https://github.com/dhylands/micropython/blob/teensy-usb/teensy/uart.c#L72
and here:
https://github.com/dhylands/micropython/blob/teensy-usb/teensy/uart.c#L272-L287

rolandvs, 2016-11-26:
Dear Dave,

Alright this seems a clean approach, however not yet in the current micropython master code for the STMHAL?
Also I get a bit confused seeing both machine and pyb with different implementations for I2C, etc.

dhylands, 2016-11-26:
I think what @dpgeorge was hinting at was that you could initialize it to use CTS/RTS on the wrong pins, and then once the uart is initialized, use the Pin.init function to set the alternate function up the way you wanted. This should work if you're looking for something *now* which could be done purely in python.

dpgeorge, 2016-12-05:
Commit 27a503f63284a1751f20d7f5295690fec5e1c462 changed the way uart pins are configured to allow full customisation of port and pin number.

@rolandvs you should now be able to easily implement your board config.

closed 2016-12-05 by dpgeorge

rolandvs, 2016-12-09:
@dpgeorge THANKS! I read about it in #2662.