micropython/micropython #2022: Cant compile on OSX

dmarkey, 2016-05-01:
CC ../lib/utils/pyhelp.c
CC ../lib/utils/printf.c
CC ../lib/fatfs/ff.c
CC ../lib/fatfs/option/ccsbcs.c
LINK build/firmware.elf
../lib/axtls/_stage/libaxtls.a: error adding symbols: Archive has no index; run ranlib to add one
make: *** [build/firmware.elf] Error 1

pfalcon, 2016-05-01:
This is very strange, because the same toolchain works on Linux. You would need to provide all the information what/how you're doing things.

dmarkey, 2016-05-01:
So esp-open-sdk with sed -i.bak '/__need_size_t/d' ./crosstool-NG/.build/src/gmp-5.1.3/gmp-h.in patch to compile properly

Then just followed instructions...
git clone
git submodule update --init
cd esp8266
make axtls
make

pfalcon, 2016-05-01:
Well, I pushed update to axtls submodule which should fix this, though I'm still not sure why it happens to the toolchain built on OSX and doesn't happen to the same toolchain built on Linux.

dmarkey, 2016-05-01:
No cigar,
t-section-literals -mforce-l32 -DLWIP_OPEN_SRC -Dabort=abort_ -DRT_MAX_PLAIN_LENGTH=1024 -DRT_EXTRA=3072 -Wall -Wstrict-prototypes -Wshadow -g -Os -ffunction-sections -fdata-sections -fno-unwind-tables -fno-asynchronous-unwind-tables     -c -o x509.o x509.c
x509.c: In function 'x509_new':
x509.c:77:13: warning: variable 'bi_ctx' set but not used [-Wunused-but-set-variable]
     BI_CTX *bi_ctx;
             ^
ar -rcs .././_stage/libaxtls.a ../crypto/aes.o ../crypto/bigint.o ../crypto/crypto_misc.o ../crypto/hmac.o ../crypto/md5.o ../crypto/rc4.o ../crypto/rsa.o ../crypto/sha1.o ../crypto/sha256.o ../crypto/sha384.o ../crypto/sha512.o asn1.o gen_cert.o loader.o openssl.o os_port.o p12.o tls1.o tls1_svr.o tls1_clnt.o x509.o
warning: /Library/Developer/CommandLineTools/usr/bin/ranlib: warning for library: .././_stage/libaxtls.a the table of contents is empty (no object file members in the library define global symbols)

Thats from the make axtls and:

LINK build/firmware.elf
build/py/../extmod/moduhashlib.o:(.text.sha1_update+0x0): undefined reference to `SHA1_Update'
build/py/../extmod/moduhashlib.o: In function `sha1_update':
moduhashlib.c:(.text.sha1_update+0x1f): undefined reference to `SHA1_Update'
build/py/../extmod/moduhashlib.o:(.text.sha1_digest+0x0): undefined reference to `SHA1_Final'
build/py/../extmod/moduhashlib.o: In function `sha1_digest':
moduhashlib.c:(.text.sha1_digest+0x1a): undefined reference to `SHA1_Final'
build/py/../extmod/moduhashlib.o:(.text.sha1_make_new+0x0): undefined reference to `SHA1_Init'
build/py/../extmod/moduhashlib.o: In function `sha1_make_new':
moduhashlib.c:(.text.sha1_make_new+0x32): undefined reference to `SHA1_Init'
build/py/../extmod/modussl.o:(.text.mod_ssl_wrap_socket+0x10): undefined reference to `ssl_ctx_new'
build/py/../extmod/modussl.o:(.text.mod_ssl_wrap_socket+0x14): undefined reference to `ssl_server_new'
build/py/../extmod/modussl.o:(.text.mod_ssl_wrap_socket+0x18): undefined reference to `ssl_client_new'
build/py/../extmod/modussl.o:(.text.mod_ssl_wrap_socket+0x1c): undefined reference to `ssl_handshake_status'
build/py/../extmod/modussl.o:(.text.mod_ssl_wrap_socket+0x20): undefined reference to `ssl_display_error'
build/py/../extmod/modussl.o: In function `mod_ssl_wrap_socket':
modussl.c:(.text.mod_ssl_wrap_socket+0x6a): undefined reference to `ssl_ctx_new'
modussl.c:(.text.mod_ssl_wrap_socket+0x80): undefined reference to `ssl_server_new'
modussl.c:(.text.mod_ssl_wrap_socket+0x91): undefined reference to `ssl_client_new'
modussl.c:(.text.mod_ssl_wrap_socket+0x99): undefined reference to `ssl_handshake_status'
modussl.c:(.text.mod_ssl_wrap_socket+0xad): undefined reference to `ssl_display_error'
build/py/../extmod/modussl.o:(.text.socket_close+0x0): undefined reference to `ssl_free'
build/py/../extmod/modussl.o:(.text.socket_close+0x4): undefined reference to `ssl_ctx_free'
build/py/../extmod/modussl.o: In function `socket_close':
modussl.c:(.text.socket_close+0x15): undefined reference to `ssl_free'
modussl.c:(.text.socket_close+0x1d): undefined reference to `ssl_ctx_free'
build/py/../extmod/modussl.o:(.text.socket_write+0x0): undefined reference to `ssl_write'
build/py/../extmod/modussl.o: In function `socket_write':
modussl.c:(.text.socket_write+0xf): undefined reference to `ssl_write'
build/py/../extmod/modussl.o:(.text.socket_read+0x0): undefined reference to `ssl_read'
build/py/../extmod/modussl.o: In function `socket_read':
modussl.c:(.text.socket_read+0x26): undefined reference to `ssl_read'
make: *** [build/firmware.elf] Error 1

pfalcon, 2016-05-01:
Ah, I see, it's the same issue as was previously reported for lwip - system incompatible ar is used. Fixed in master.

closed 2016-05-01 by pfalcon