micropython/micropython #2028: unix: super() appears not to work with multiple inheritance

peterhinch, 2016-05-03:
This derives from [this forum post](http://forum.micropython.org/viewtopic.php?f=2&t=1842&p=10418#p10418).
```python
class A:
    def __init__(self):
        print("A.__init__")

class B(A):
    def __init__(self):
        print("B.__init__")
        super().__init__()

class C(A):
    def __init__(self):
        print("C.__init__")
        super().__init__()

class D(B,C):
    def __init__(self):
        print("D.__init__")
        super().__init__()
```
If you issue D() at the REPL you get:
```
>>> D()
D.__init__
B.__init__
A.__init__
<D object at aac31180>
```
whereas cPython gives:
```
>>> D()
D.__init__
B.__init__
C.__init__
A.__init__
<__main__.D object at 0x7f40ecd1a358>
>>> 
```
A further issue with super() - with single inheritance - is this. Despite numerous successful uses I wrote one class where super() returned an empty string. Unfortunately any attempt to produce a test case failed. I appreciate that's a lousy bug report but I thought it worth mentioning if the code is being reviewed.

pfalcon, 2016-05-03:
The example above is a [diamond problem](https://en.wikipedia.org/wiki/Multiple_inheritance#The_diamond_problem), a case which is usually used to show how multiple inheritance sucks and a motivation to not support multiple inheritance in some languages. It's an artificial problem the practical solution of which is usually avoiding diamond pattern. As it stands, it's someone researching how MicroPython handles abstract problem of diamond pattern, they're welcome to draw obvious conclusions.

peterhinch, 2016-05-04:
I agree that multiple inheritance sucks, however I'm not convinced this is an instance of the diamond problem. The omission of the 'C' constructor doesn't seem to fit that pattern. This is confirmed by
```python
class A:
    def __init__(self):
        print("A.__init__")

class E:
    def __init__(self):
        print("E.__init__")

class B(E):
    def __init__(self):
        print("B.__init__")
        super().__init__()

class C(A):
    def __init__(self):
        print("C.__init__")
        super().__init__()

class D(B,C):
    def __init__(self):
        print("D.__init__")
        super().__init__()
```
This yields
```
>>> D()
D.__init__
B.__init__
E.__init__
<D object at 721cd1a0>
>>> 
```

hardkrash, 2016-05-08:
I would argue that python multiple inheritance does not suck.
I use mix-in classes a lot.  They are great for composting nifty features and continuing the DRY principal, and eases cleaning up bugs in those features.

Multiple inheritance enables mix-ins and can even make the Diamond pattern go away.
 
```super()``` is how you access a function from the next class into the MRO of the original class.

pfalcon, 2016-05-08:
Exactly, mixins is the common pattern how multiple inheritance used in Python, and that works well with MicroPython - if you want to add some methods to your class using mixin just go ahead.