micropython/micropython #2029: esp8266:  'Pin' object has no attribute 'PULL_NONE'

azmodie, 2016-05-04:
Traceback (most recent call last):
  File "<stdin>", line 8, in <module>
  File "neopixel", line 11, in __init__
AttributeError: 'Pin' object has no attribute 'PULL_NONE'

self.pin.init(pin.OUT, pin.PULL_NONE)
commit 8c3b5526aee160f5dd7d836f4666a874e4c3212e

should be 

self.pin.init(pin.OUT, pin.PULL_NONE)

seastate, 2016-05-05:
Similar seems to be true of class OneWire in onewire.py from esp8266/scripts.

Replacing line 16
        self.pin.init(pin.OPEN_DRAIN, pin.PULL_NONE)
with 
        self.pin.init(pin.OPEN_DRAIN)
seems to work, at least for one DS18B20.  I've not yet tried for an array of these sensors.

commit: eb54e4d065a7cbc26dc4c8dc2812d1cabdcd0c50

pfalcon, 2016-05-05:
Fixed in df2b1a4758500d3ac538b30fad2e5a48126ea4a0, thanks for report.

closed 2016-05-05 by pfalcon