micropython/micropython #2037: REPL input from code

Ivoah, 2016-05-06:
Is there a way to send a string to the REPL through code? Like nodemcu's `node.input()` function? It would be useful for making a REPL available over telnet or something.

dhylands, 2016-05-06:
If you running on pyboard A and you want to take some python text and execute it on pyboard A you can use eval, which is essentially what the REPL does.

Ivoah, 2016-05-06:
I'm thinking of using it on the ESP8266. Also, eval doesn't support multi-line expressions, like the REPL does. `node.input()` does support multi-line expressions
> Submits a string to the Lua interpreter. Similar to pcall(loadstring(str)), but without the single-line limitation.

dhylands, 2016-05-06:
Perhaps exec then? Putting this into a file and running it (with the unix version):
```
exec("""def foo():
    for i in range(5):
        print(i)
foo()
""")
```
produces the expected output of:
```
0
1
2
3
4
```

Ivoah, 2016-05-06:
That still requires all of the input to be in one string. I believe with `node.input()` it is possible to do something like this:
```lua
node.input("function foo()")
node.input("for i=0, 5 do")
node.input("print(i)")
node.input("end")
node.input("end")
node.input("foo()")
```
It would be possible to make a pure python solution using `exec()` though, just have to check if the expression is finished before sending it to `exec()`

dhylands, 2016-05-06:
The repl code (written in C) is written in a way that it can be called a character at a time. So I guess it should be possible to make a python callable function which passes a character at a time to a repl object. I don't think that anybody has done this yet though.

Ivoah, 2016-05-06:
Okay, great! I'll poke around and see what I can do.

Ivoah, 2016-05-06:
So after some initial poking around, it looks like ['int readline_process_char(int c)'](https://github.com/micropython/micropython/blob/b8945517723efaafa63df9ce52bb9d3fa3fceda5/lib/mp-readline/readline.c#L101) is what I want to hook into.

pfalcon, 2016-05-06:
REPL is already available over WebREPL. Similar mechanism can be used for anything.

closed 2016-05-06 by pfalcon

Ivoah, 2016-05-07:
> REPL is already available over WebREPL. Similar mechanism can be used for anything.
after looking at the webrepl source, it seems like `uos.dupterm` is what I'm looking for

dragon788, 2017-04-14:
I just ran across this today, but the microPython interactive REPL has "paste mode" Ctrl+E and "raw mode" Ctrl+A which allows you to paste in a string and only run it after hitting Ctrl+D. Perhaps this could be hooked in a similar manner?