micropython/micropython #2040: esp8266: missing UART.deinit()

deshipu, 2016-05-07:
I assume that since there has been a release, discrepancies with the documentation are actually bugs now. Please let me know if that's not a case, then I will stop submitting them.

The documentation at http://micropython.org/resources/docs/en/latest/esp8266/library/machine.UART.html#uart.deinit mentions a ``UART.deinit()`` method, which doesn't seem to be present on the ESP8266 port:

```
>>> UART.deinit()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: type object 'UART' has no attribute 'deinit'
```

deshipu, 2016-05-07:
I also seems to be missing ``any`` and ``sendbreak``.

pfalcon, 2016-05-07:
Yes, they are bugs. It's unclear how to deal with them in the meantime though - by pruning from documentation or by having these issues open potentially for prolonged time. The best solution is that people who need start to dig into that (not necessarily provide ready-made patches, but e.g. start to discuss how to implement those methods on esp8266 and what they should do: esp8266 can't really work real-world'ly without terminal on UART).

deshipu, 2016-05-07:
How about I submit a pull request for conditional removal of those particular methods from the ESP8266 port documentation? We can always add them later if/when they get implemented?

I noticed the ADC section is completely off, too, that should probably get an ESP8266-only version.

pfalcon, 2016-05-07:
> I noticed the ADC section is completely off, too, that should probably get an ESP8266-only version.

I noticed that too, but the usual question which direction it should change. Mind that we implement consistent and portable hardware API per https://github.com/micropython/micropython/wiki/Hardware-API (but that doc itself may need changes to support portable and consistent API). All in all, the current actual info for esp8266 is in it's quickref doc.

roger-, 2017-05-06:
Any updates on this? I'm trying to reuse the UART with the webrepl and  deinit might help.

sxwid, 2020-10-06:
> How about I submit a pull request for conditional removal of those particular methods from the ESP8266 port documentation? 
>
Is there any progress on these methods over 3 years later? In case not I support the ask for removal in the documentation - very annoying! Whats an argument to leave unimplemented methods documented???

closed 2021-04-02 by deshipu