micropython/micropython #2041: Pin Interrupts don't appear to take keyword arguments

slzatz, 2016-05-07:
Documentation for esp at http://docs.micropython.org/en/latest/esp8266/esp8266/tutorial/pins.html

reads

    >>> p0.irq(trigger=Pin.IRQ_FALLING, handler=callback)
    >>> p2.irq(trigger=Pin.IRQ_RISING | Pin.IRQ_FALLING, handler=callback)

However, the irq method doesn't take keyword arguments but works fine with positional arguments.

dpgeorge, 2016-05-08:
Which version of the SW did you try?  Keyword args were implemented in 8e130fcf2b4b27b0e97f87bae3bff2b9051f1955.

slzatz, 2016-05-08:
Sorry for the noise - was testing build right before that and didn't see that change.  Will make sure I am on the latest version for any future comments.

closed 2016-05-08 by slzatz