micropython/micropython #2043: 512MB ESP8266

Ivoah, 2016-05-07:
I know that the 512MB ESP8266 chips don't have a filesystem, but would it be possible for them to have a single-file filesystem? Like only make boot.py available, so that it's still possible to put permanent code on them

pfalcon, 2016-05-07:
I'm afraid no, we feature-pack esp8266 build so the actual code takes close to 512K. However, if for some reason you're interested in that, you can: 1) figure out what's the minimum filesystem size with 4K sectors (current limit enforced is based on 512K sectors); 2) try to prune port config to free up that much of flash.

Another approach would be to integrate another filesystem (but the above is much more low-hanging fruit).

dpgeorge, 2016-12-02:
Support for 512k modules is added via "make 512k" target.  More recent issue discussing this is #2557.

closed 2016-12-02 by dpgeorge