micropython/micropython #2044: PYB os.mkfs() fails to execute

noelhenson, 2016-05-07:
No success in executing os.mkfs()
Versions 1.6, 1.7 1.8
Documentation shows a string path parameter is needed.
Experimentation shows two parameters are needed.
Repro:

>>> import os
>>> os.mkfs()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function missing 2 required positional arguments
>>> os.mkfs('/flash')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function missing 1 required positional arguments
>>> os.mkfs('/flash','PYBFLASH')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'str' object has no attribute 'readblocks'
>>>

dpgeorge, 2016-05-11:
You need to pass the block device and a mount point:
```
os.mkfs(blockdevice, '/mnt')
```
In the case of the internal flash the block device is `pyb.Flash()` and you can format it using:
```
flash = pyb.Flash()
os.umount('/flash') # unmount first
os.mkfs(flash, '/flash')
os.mount(flash, '/flash') # mount newly formatted filesystem
```

cpranzl, 2016-08-11:
No success in executing os.mkfs()
Version: 1.8.2-9-g805c2b9
Board: WeMos D1 Mini

Repro:
>>> import os
>>> os.mkfs()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'mkfs'

noelhenson, 2016-08-11:
>             File "", line 1, in

You need to pass the block device and a mount point:

|os.mkfs(blockdevice, '/mnt') |

dpgeorge, 2017-06-14:
os.mkfs() has now been removed from the stmhal port.  Instead one must use os.VfsFat.mkfs(bdev).

closed 2017-06-14 by dpgeorge