micropython/micropython #2052: esp8266: can't time pulses with good accuracy

roger-, 2016-05-08:
I'm testing some of the approaches in issue #570 to timing a pulse with the `time.ticks_us()` function, but I can't get better than ~60 us resolution (half if I double the clock speed):

```python
>>> x=[time.ticks_us() for i in range(10)]
>>> [x[i] - x[i-1] for i in range(1,10)]
[90, 22, 65, 65, 45, 66, 65, 22, 83]
```

It seems that the hardware/RTOS timers only have 1 ms resolution, so is there another way to do this? I'm looking to get something like <10 us resolution.

(BTW I think it'd be useful to have a standard/reliable approach to pulse timing on each port, like the Arduino `pulseIn()` function.)

dpgeorge, 2016-05-11:
Doing it in pure Python you won't be able to get better than 100us or so resolution.  You'd need either a custom C function (like pulseIn) or to write it in inline assembler or viper (which are not yet supported).

roger-, 2016-05-11:
Ah okay, thanks. I'll wait for viper then. 

A `Pin.time_pulse()` method would be really useful to add at some point!

pfalcon, 2016-05-11:
I think we should add method(s) to time pulses. So, initial RFC on how it should be named, what it should do, comparing with existing popular APIs is welcome. For starters, I don't think it should be a Pin method, but a standalone function.

pfalcon, 2016-05-11:
Well, 1-Wire driver already would be doing something like that, so RFC should cover how it would be refactored to take advantage of new API call.

roger-, 2016-05-11:
The Arduino function is defined as `pulseIn(pin, value, timeout)`. Since it always requires a pin I'm inclined to say it should be a method.

pfalcon, 2016-05-11:
That means each pin implementation should define it. But implementation is always the same, but we won't be bloating it with introduction of base class, so independent function is the obvious design choice.

dpgeorge, 2016-05-11:
> But implementation is always the same

You mean the same across pins, or across ports?  Are you talking about the C level or Python level of implementation?

If it's only ever useful for timing pins then it should be a pin method.

pfalcon, 2016-05-11:
Across ports. C-level.

> If it's only ever useful for timing pins then it should be a pin method.

That's word-playing. It's an operation. Operations are not methods. Cf. `len()` vs underlying method `__len__()`. (Here, operation time_pulse_ms() vs underlying method .value()).

roger-, 2016-05-11:
I notice that `pulseIn()` waits for a rising (falling) edge and times until the falling (rising) edge. It might be useful to have the option to start timing right away so pulses that are already active can be measured. 

So a possible signature could be: `time_pulse_us(pin, state, timeout=1000000, wait=False)`, where `wait` decides if it should wait for the edge trigger to start.

pfalcon, 2016-06-05:
Fixed with https://github.com/micropython/micropython/pull/2091

closed 2016-06-05 by pfalcon