micropython/micropython #2053: slice object passed to __setitem__/__getitem__ seems unuseable

PinkInk, 2016-05-08:
Hi, the slice object passed into `__setitem__` and `__getitem__` of a user defined class seems not to be useable or, at least, far from an intuitive derivative of cpython behaviour ... examples;

```python
>>> class A:
...     def __getitem__(self,idx):
...         print(idx)
...         print(type(idx))
...         print(idx.start)
...
>>> A()[1:2:3]
slice(1, 2, 3)
<class 'slice'>
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 5, in __getitem__
AttributeError: no such attribute
>>> class A:
...     def __getitem__(self,idx):
...         print(idx[0])
...
>>> A()[1:2:3]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 3, in __getitem__
TypeError: object is not subscriptable
>>> class A:
...     def __getitem__(self,idx):
...         print(type(idx) is slice)
...
>>> A()[1:2:3]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 3, in __getitem__
NameError: name not defined
```

pfalcon, 2016-05-08:
Nobody needed it for compelling reason, that is.

PinkInk, 2016-05-08:
OK; suspected that might be the case ... not sure that I'd argue the case for my reason being compelling, in particular if no-one's asked for it before, there are doubtless other ways, I'll rethink.

pfalcon, 2016-05-08:
> if no-one's asked for it before

You can search this tracker and see. Until people start to do that, get together, and argue together, it only proves there's no compelling reason to have it ;-).

PinkInk, 2016-05-08:
OK, agree; for the record, whilst closing - in case additional demand ever causes this request to resurface - I 'desire' it for the purposes of a reusable python class to minimize generation of intermediary objects when working on binary data, by 'sliding data around' inside a larger buffer rather than joining multiple small objects into larger ones, and then chopping them up again and doing it again (hope this makes sense ;o)

Sort of akin to a limited-scope version of what you describe in this post http://forum.micropython.org/viewtopic.php?f=3&t=103&p=351&hilit=slice#p351 'way back when'.

The extremely limited resources available on the esp port make the ability to achieve similar unsightly shenanigans more compelling.  In my mind.

closed 2016-05-08 by PinkInk