micropython/micropython #2054: esp8266: WebREPL daemon leaks WiFi RX buffers

pfalcon, 2016-05-08:
A known issue from 1.8 release.

pfalcon, 2016-05-08:
I added webrepl.start_foreground(). With it, a leak doesn't occur. So, it's confirmed that "daemon mode", specifically accept callback, is the culprit.

pfalcon, 2016-05-15:
The problem would be accept callback calling back lwIP recursively via ets_loop_iter(). Not confirmed, just a hunch.

pfalcon, 2016-05-15:
> The problem would be accept callback calling back lwIP recursively via ets_loop_iter().

Kind of. The actual problem was a bit different than what I imagined initially. And @dpgeorge as usual was right asking: "How does it work at all?" A: It doesn't. lwIP isn't reenterable, uses global variables to track processing state of the current packet, etc. Accept callback is called when ACK packet from peer is received, and it's not supposed to make any packets be "received" while it works, and sending any packets will cause ACKs received. Accept callback is there to just decide whether connection can be accepted or not. So, we can't call any elaborated callback from it. Changing that in lwIP codebase is problematic and not right (it should be called early to decide whether to drop conn at all). So, accept callback can just schedule something to be done later. There's "poll" PCB callback which is called when lwIP isn't inside receiving/sending tasks. I decided to use it. But it's called very rarely (resolution is ~0.5s), so had to patch lwIP still to call it soon after accept callback which schedules it.

pfalcon, 2016-05-15:
Fix pushed to master as afce978aca77805e84a2442c93c26dab045784f8. Requires esp-open-sdk update per https://github.com/micropython/micropython/issues/2066#issuecomment-219306071

dpgeorge, 2016-05-16:
I tried this out using the latest esp-open-sdk.  I can confirm that there are no longer any OS error messages.  But nevertheless webrepl still doesn't allow more than 5 consecutive connections.  Using `webrepl.start(password='')` I can connect exactly 5 times in a row using either webrepl.html or webrepl_cli.py (ie REPL or file transfer) but the 6th attempt doesn't do anything, there is no response from the board (client is not asking for password).

Using `webrepl.start_foreground()` I can connect many times (tried up to 10 times in a row with webrepl_cli.py, no problems).

pfalcon, 2016-05-16:
What does esp.esf_free_bufs(4) say you? I verified it stays at 7 after connect, and stay such over consecutive reconnects. I didn't try to connect 5+ times in row (more like 3), will try it.

dpgeorge, 2016-05-16:
> What does esp.esf_free_bufs(4) say you?

It stays at 7.  But 6th attempt at connect still fails.

pfalcon, 2016-05-16:
LOL, there's just some dirty magic stuck to this number and webrepl. Confirmed. Specifically, connection to webrepl hangs on 6th attempt on client side, and nothing is reported on server side, i.e. lwip just doesn't accept new connection (doesn't send ACK to SYN). The workaround is to webrepl.stop() and .start(), but after that, it appears to be able to handle just 1 connect, 2nd leads to hang (ditto for the following restarts).

pfalcon, 2016-05-16:
Here would be it:
~~~
>>> socket.print_pcbs()
Active PCB states:
Local port 8266, foreign port 40598 snd_nxt 29716 rcv_nxt 109179712 State: CLOSE_WAIT
Local port 8266, foreign port 40505 snd_nxt 9062 rcv_nxt -1695479457 State: CLOSE_WAIT
Local port 8266, foreign port 40504 snd_nxt 8208 rcv_nxt -1724653079 State: CLOSE_WAIT
Local port 8266, foreign port 40498 snd_nxt 7485 rcv_nxt -867479100 State: CLOSE_WAIT
Local port 8266, foreign port 40483 snd_nxt 9592 rcv_nxt 1089215787 State: CLOSE_WAIT
Listen PCB states:
~~~

lwIP config supports 5 active sockets, and webrepl doesn't close sockets on which disconnect received.  To fix that, would need to add way to have dupterm subsystem to nofity dupterm object on connection closure. A perhaps surprising way to do that is to call .close() on dupterm object. Do we want to extend dupterm protocol in such way?

deshipu, 2016-05-16:
With the most recent head (afce978aca77805e84a2442c93c26dab045784f8) I can't seem to be able to connect to the webrepl.

The procedure:
```
cd [...]/esp-open-sdk
make clean
git pull
make
[...]
cd [...]/micropython/esp8266
make clean
git pull
git submodule update
make -j4
[...]/esptool.py --port /dev/ttyUSB0 erase_flash
[...]/esptool.py --port /dev/ttyUSB0 --baud 460800 write_flash --flash_size=8m -fm dio 0 build/firmware-combined.bin
```

Then in the REPL over serial:
```
rll��|�l�|�l�b|����r�b�b��nn�lnn���bp��lrlrlp�n��lbn�|���p�b��nn�l��l`�nnl`nr���n�bl`p�n�r�����bn�|l�b��nn�l`�nnl`nr���n�l`p�n�r�������bn�|���r�b��nn�l`�nnl�l`nr���n�l`r��n�l`�l��|��rrnb��l��b�lb쌜���l�|�b�lrldon't use rtc mem data
ll��|��rrnb��lll��b�b쌜����b��lrl�l��|��rrnb��l���b�b쌜���b�lblrl��rl���bl�br|�b�������l��n�nn�ll���l��l����ll`�n����b��b���bl�b�lrlr�n�����n����bpb������r�쎜�lbl��ln�Bootloader Flash size appear to have been set incorrectly, trying to fix
bcn 0
del if1
usl

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 30616, room 16
tail 8
chksum 0xd2
load 0x3ffe8000, len 1048, room 0
tail 8
chksum 0xd8
load 0x3ffe8420, len 2652, room 0
tail 12
chksum 0x13
csum 0x13
�l��|��rrnb��l��b�lb쌜���l�|�b�lrldon't use rtc mem data
ll��|��rrnb��lll��b�b쌜����b��lrl�l��|��rrnb��l���b�b쌜���b�lblrl��rl���bl�br|�b�������l��n�nn�ll���l��l����ll`�n����b��b���bl�b�lrlr�n�����n����bpb������r�쎜�lbl��ln�Performing initial setup
bcn 0
del if1
usl
add if1
#4 ets_task(4020e20c, 29, 3fff6f10, 10)
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
could not open file 'main.py' for reading

#5 ets_task(401002fc, 3, 3fff6190, 4)
MicroPython v1.8-77-gafce978 on 2016-05-16; ESP module with ESP8266
Type "help()" for more information.
>>> import network
>>> network.WLAN(network.AP_IF).active(False)
bcn 0
del if1
usl
mode : null
>>> wlan = network.WLAN()
>>> wlan.active(True)
mode : sta(18:fe:34:d3:7f:a4)
#6 ets_task(4020e1d4, 28, 3fff6b50, 10)
add if0
>>> wlan.connect("swamp", "[...]")
>>> scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt

connected with swamp, channel 1
dhcp client start...
ip:192.168.0.13,mask:255.255.255.0,gw:192.168.0.1

>>> import pm open,type:2 0
webrepl
>>> webrepl.start()
WebREPL daemon started on ws://192.168.0.13:8266
Started webrepl in setup mode
>>> import port_diag
Flash ID: 1640e0 (Vendor: e0 Device: 4016)
Flash bootloader data:
Byte @2: 02
Byte @3: 40 (Flash size: 4MB Flash freq: 40MHZ)
```

Now, when I try connecting with the web-based webrepl client (Chromim Version 49.0.2623.108 Ubuntu 15.10 (64-bit)), I get this on the serial:

```

First-time WebREPL connection has been received. WebREPL initial setup
will now start over this connection. During setup, UART REPL will be
non-responsive. After setup finishes, the board will be rebooted. In
case of error during setup, current session will continue.

If you receive this message unexpectedly, it may mean that your WebREPL
connection is being hacked (power off board if unsure).
```

And the serial REPL is unresponsive (pressing enter doesn't make ">>>" appear.

EDIT: After about 1 minute, the password prompt appears and I can set the new password, after which the module restarts. Then, when I try to connect with web client, I get:

```
WebREPL connection from: ('192.168.0.12', 47072)
```
And the serial REPL is again unresponsive. After about a minute I get the password prompt and the web client works fine. The serial REPL also becomes responsive and receives all the buffered enters.

The ``webrepl_cli.py`` also makes Micropython freeze, but it never connects -- it stays frozen forever.

pfalcon, 2016-05-16:
> Chromium

Please try Firefox. Chromium is known to be aggressive with multiple parallel requests. I'll test Chromium later (didn't do that after these changes).

> lwIP PCBs:
> >>> pm open,type:2 0

If you get that in UART REPL (that's where it should be run), it means that you didn't rebuild lwip in esp-open-sdk completely. I'd suggest re-checking git revision you have and that you have it cleaned, recompiled, and installed (including headers). The output which should be there is really useful for diagnosing high-level networking issues. (But feel free to skip it until after I try Chromium myself.)

pfalcon, 2016-05-16:
Btw, from yesterday, I see that even Firefox makes 2 parallel connection when "Connect" is pressed in webrepl client - not every time, but with high probability. Never saw that before %). And no, it's not related to the change committed, it started before (so I had pretty hellish debugging session, wading thru few unrelated issues).

pfalcon, 2016-05-16:
Sorry, can't reproduce with "Chromium Version 49.0.2623.108 Ubuntu 14.04 (64-bit)", i.e. exactly the same version as yours, and already initialized webrepl - everything is instantaneous for me.

> WebREPL daemon started on ws://192.168.0.13:8266

This would mean that you connect via router? Please try esp8266's AP directly to rule out router issues.

deshipu, 2016-05-16:
On Firefox it takes shorter -- about 20s. The command-line file transfer utility doesn't work at all, though.

If "make clean" doesn't let me build fresh code, then how should I do it? I certainly won't be doing that today -- just doing "make clean" on the esp-open-sdk takes a good half an hour on my computer. Rebuilding it takes another 45 minutes.

I will try the AP interface.

pfalcon, 2016-05-16:
From esp-open-sdk dir:

~~~
cd esp-open-lwip
git log
~~~

What's the top rev you see? Then go to https://github.com/pfalcon/esp-open-sdk and see what current repo has for that submodule. If revs don't match, you didn't update submodules as described in esp-open-sdk's README. If you did update, then git submodule --help and see what flags are missing in the docs and please report back (and yes, that's why people hate git submodules).

> just doing "make clean" on the esp-open-sdk takes a good half an hour on my computer. Rebuilding it takes another 45 minutes.

As an advanced user, you don't need to rebuild crosstool-NG part of it in 95% of cases, there're other clean-* targets in the makefile.

deshipu, 2016-05-16:
OK, mystery solved, last commit in my esp-open-sdk is from December last year. A wonder it even worked at all. Rebuilding now.

pfalcon, 2016-05-16:
Ok, for completeness, I retested with post-erase_flash case, works well for me too.

dpgeorge, 2016-05-17:
> To fix that, would need to add way to have dupterm subsystem to nofity dupterm object on connection closure. A perhaps surprising way to do that is to call .close() on dupterm object. Do we want to extend dupterm protocol in such way?

That's one way to do it [to be clear, a dupterm object must then provide read, write and close methods].

Another way would be to specify: if the dupterm object's read() method returns a str/bytes object of zero length then the dupterm will be deactivated and it's up to the object to close itself before it returns the zero-length str/bytes object.

What about if read/write raise an exception?  That also deactivates the duplicated term, so should the object close before it raises, or should the subsystem call close() if an exception is raised?

pfalcon, 2016-05-17:
> to be clear, a dupterm object must then provide read, write and close methods

Yes, which is fair assumption given that dupterm object is a stream object.

> if the dupterm object's read() method returns a str/bytes object of zero length then the dupterm will be deactivated and it's up to the object to close itself before it returns the zero-length str/bytes object.

That's why I wrote that making dupterm subsystem call .close() is surprising[ly simple] way to not induce such complications. Nothing like that is required from a stream object, and inducing rules like the above means that dupterm requires something more than a normal stream object.

> What about if read/write raise an exception? That also deactivates the duplicated term, so should the object close before it raises, or should the subsystem call close() if an exception is raised?

From the above the answer should come natural - stream object doesn't manage its own lifecycle, it's managed externally by the client. If dupterm requires particular lifecycle management, it implements it on its side, instead of require each dupterm-able object take care of that.

dpgeorge, 2016-05-17:
Ok, so we agree that close() should be added to the spec for a dupterm-able entity, and it will be called by the subsystem when appropriate.

pfalcon, 2016-05-17:
Ok, will implement that.

pfalcon, 2016-05-17:
> Accept callback is there to just decide whether connection can be accepted or not. So, we can't call any elaborated callback from it. Changing that in lwIP codebase is problematic and not right (it [...]

I'd elaborate on this, and why we're hitting this error with lwIP. It's because we're trying to do blocking calls in accept handler. lwIP is not normally designed to work like that at all (well, that API which we use isn't designed to do blocking calls at all). How it would be done in normal lwIP way is that accept callback installs recv callback which then does async processing. We can't do that in our case because we have websocket handshaking functionality separated out from websocket object. It first of all does synchronous handshake processing, and then there would not be a way to handover further communication to actual websocket object. That shows how good such design from "lwIP daemon" support point of view - it's not. It's certainly nice, simple and extensible from standard BSD Socket API point of view, but if we need to support daemon functionality, we need to go beyond that and play by lwIP rules, and the rule is simple: there's a single callback. So, typical callback composability issues show up, etc. So, would need to move handshake methods to websocket object after all, and make the whole connection process to (be able to) work using .read() and asynchronously. No hurry with that of course, as workaround was found (which requires to patch lwip to remind).

pfalcon, 2016-05-17:
Now another aspect of it: if there was such issue with accept callback, then what about recv callback. It may seem then it's good: reading from it happens in async manner. But. When enough data is received, it's processed and reply is sent in blocking way (i.e. webrepl socket is non-blocking for recv and blocking for send). I already shared my surprise and paradigmatical aspect of it - that's required for natural flow control. Even a simple input (Python program) may produce large output, megabytes, or just infinite. There's nowhere to buffer such large output. Small amounts, can be buffered in socket send buffer, and lwip does that automatically. But as soon as it get exhausted, the only way to send more data is by having sent the previous data. So, we block and wait for it to be sent. This part, can't be easily done in another way, like websocket handshake issue discussed above (and "easily" is a wrong word here - writing HTTP async parser in C is pretty boring, that's why there's a sync Python one).

To handle that, there should be complete coroutine-based design for both REPL input and output. We had push-based, callback design for REPL input in esp8266 initially, but much more would be required for handling both input and output. We had troubles maintaining even just input callback design, and it would be a big heck to write input/output coro design, and then we'd need to switch all ports to it, because maintaining such implementation in parallel with "normal" would be quite cumbersome.

So, again, there's no easy solution to blocking sends, but doing them from a recv callback is potentially as unsafe as from an accept callback. There's the same processing paths, the same global variables, and actually the behavior we experience, e.g. gradual slowdowns down to a deadlock are pretty explainable by "packet losses" caused by global variable overwriting. So, I'm going to try to move Python recv callback calling to lwip "poll" callback just like the accept has been.

dpgeorge, 2016-05-17:
> So, I'm going to try to move Python recv callback calling to lwip "poll" callback just like the accept has been.

Ok.

Alternatively, the scheduling of the callback can be done at the ets post level (posting a notification that a task needs to be run), or even using ets_event_poll, making it check for any pending callbacks that need calling.

pfalcon, 2016-05-17:
Well, we shouldn't make this esp-specific, it should be handled on lwip level (even if it requires patched lwip).

dpgeorge, 2016-05-17:
> it should be handled on lwip level

But then it's lwip specific.  Eg it needs to be reimplemented for the cc3100/cc3200 driver.

pfalcon, 2016-05-17:
Well, not exactly. lwip is a leading open-source TCP/IP stack. cc3100/cc3200 *driver* should just allow integration with any TCP/IP stack (including lwip), instead of it (the driver) being a TCP/IP stack itself. Vendors who don't agree are welcome to pay for porting of webrepl to their stuff.

dpgeorge, 2016-05-17:
> cc3100/cc3200 driver should just allow integration with any TCP/IP stack (including lwip), instead of it (the driver) being a TCP/IP stack itself.

Why so?  The aim of these chips is to take the burden of a lot of the processing.

> Vendors who don't agree are welcome to pay for porting of webrepl to their stuff

No, we are the ones who have to pay because we will end up writing the code.  WiPy won't have webrepl until we implement it.  I suspect at some point we'll need to code up some lwip-independent callback mechanism.

pfalcon, 2016-05-17:
> The aim of these chips is to take the burden of a lot of the processing.

Apparently, they succeeded, and arbitrary limitations, non-extensibility, bugs, and security issues aren't big price for that.

 > WiPy won't have webrepl until we implement it.

WiPy has telnet and ftp and the rest of uPy doesn't have them, so what? WiPy has its own vendor behind it, and there're lower-hanging fruits to add to WiPy (like generic uPy VFS system instead of, again, WiPy's adhoc one).

> I suspect at some point we'll need to code up some lwip-independent callback mechanism.

We already walk a thin line between implementing our own RTOS inside uPy. I'm sure future versions of uPy will be RTOS-based (that's why esp8266 port is so precious by still zigzagging to avoid it and proving it's possible), I just hope that it won't be some RTOS with fishy strings attached (like FreeRTOS), but one which brings real value. Zephyr comes with BLE stack for example. Btw, I attribute ESP32 delay partly to BLE licensing issues - they have hardware, they don't have software to shoplift to run on it. Now they have.

pfalcon, 2016-05-18:
@deshipu : Can you confirm that rebuild solved the issue for you?

deshipu, 2016-05-18:
Yes, I already confirmed in https://github.com/micropython/micropython/issues/2054#issuecomment-219562602

pfalcon, 2016-05-18:
Sorry, missed it at that time, maybe you edited it later. Thanks.

pfalcon, 2016-05-20:
> Ok, so we agree that close() should be added to the spec for a dupterm-able entity, and it will be called by the subsystem when appropriate.

This was implemented, tested to work as expected, and pushed to master.

dpgeorge, 2016-05-23:
I can confirm that this is now fixed, at least in my testing.  I used webrepl_cli.py and webrepl.html to connect multiple times (20+) and it worked fine.

pfalcon, 2016-05-23:
@dpgeorge : Any changes to file transfer behaviors on your side? We have surprising reports of it working too well already too: http://forum.micropython.org/viewtopic.php?f=16&t=1939#p10952

dpgeorge, 2016-05-23:
> Any changes to file transfer behaviors on your side?

Yes it seems to be *much* more reliable!  I can copy 256k to/from the device.  ESP to PC goes at about 25k/s.

mordae, 2020-02-11:
@dpgeorge: Shouldn't this be closed?

dpgeorge, 2022-03-30:
Yes!  This can be closed.

closed 2022-03-30 by dpgeorge