micropython/micropython #2062: ussl module TLS SNI (Server Name Indication) support

vshymanskyy, 2016-05-10:
It looks like currently TLS SNI is not supported.
This is required to access many SSL-enabled sites and platforms - and should be quite simple to implement.
Could you look into this and check if it will be possible to get this feature?

pfalcon, 2016-05-28:
Support for SNI depends on the underlying TLS library. Upstream axTLS library that we used doesn't support SNI.

vshymanskyy, 2016-05-28:
Strange.. axTLS is also used in ESP8266 Arduino port, but they have added SNI in this commit: https://github.com/esp8266/Arduino/commit/8c65f2fcd01b5a8de0e18144be19b15075e06ebf
Hope that helps...

pfalcon, 2017-06-13:
SNI support is added via upstream axTLS 2.1.3: https://github.com/micropython/micropython/issues/3140

closed 2017-06-13 by pfalcon