micropython/micropython #2070: /esp8266/scripts: Hidden .DS_Store files cause SyntaxError on _boot.

mcauser, 2016-05-12:
I was receiving the following error on every boot when `make` was run with a `.DS_Store` file in /esp8266/scripts

```
Traceback (most recent call last):
  File "_boot", line 1
SyntaxError: invalid syntax
could not open file 'boot.py' for reading
could not open file 'main.py' for reading
#4 ets_task(401002fc, 3, 3fff6190, 4)
MicroPython v1.8-52-gbc04dc2 on 2016-05-12; ESP module with ESP8266
```

The `.DS_Store` file was making its way into `/esp8266/build/frozen.c` and affecting `/esp8266/build/firmware.*`

I cloned a 2nd copy of this repo and compiled and uploaded successfully.
The only difference being a few .DS_Store files.

Mac OS X litters the directory structure with .DS_Store files when you browse using the Finder.

Can they be ignored on `make`?

mcauser, 2016-05-12:
In /esp8266/Makefile, something like this?
before: `$(wildcard $(SCRIPTDIR)/*)`
after: `$(filter-out .DS_Store, $(wildcard $(SCRIPTDIR)/*))`

mcauser, 2016-05-12:
Forum discussion: http://forum.micropython.org/viewtopic.php?f=16&t=1898&p=10707
What is a .DS_Store file? https://en.wikipedia.org/wiki/.DS_Store

dpgeorge, 2016-05-12:
I can't reproduce this.  Putting a .DS_Store file (containing some random characters) in esp8266/scripts/ does not give any error.

mcauser, 2016-05-12:
I zipped the offender: http://whale90.bf.net.au/.DS_Store.zip
/esp8266/build/frozen.c: https://gist.github.com/mcauser/08e0f6395fa91aa05b2b4b057ac63329

dpgeorge, 2016-05-12:
Ok, confirmed.  The issue is that tools/make-frozen.py is incorrectly escaping non-ascii chars.  The C rules for parsing \x hex constants within a string are different to Python, so using repr(data) from Python does not always make the correctly escaped string to put into C.  In your case the data from DS_Store had `"\x00\x00\x00\x01Bud1\x00..."` and the `\x01B` is parsed by C as a single character, not 2 as in Python.

mcauser, 2016-05-13:
I use this alias in my bash profile which nukes all .DS_Store files in the current dir and below.
`alias rmds='find . -name .DS_Store -type f -print -delete'`

dpgeorge, 2016-05-13:
Should be fixed by 5985e41afca9d440dd932ab512c23dbc1e002499.

closed 2016-05-13 by dpgeorge

mcauser, 2016-05-13:
It works!

klapauciusisgreat, 2017-03-30:
Did this problem recur in the latest 1.8.7 release ? Specifically, esp8266-20170108-v1.8.7.bin ? Same symptoms on a macbookpro, and I got esp8266-ota-20170330-v1.8.7-526-g29dd92c.bin working without problems.

klapauciusisgreat, 2017-03-30:
Never mind. What set me straight was calling esptool with erase_flash and then flashing again from scratch. Though, I still don't fundamentally grok how a firmware is cross compiled in the first place.

dpgeorge, 2017-03-31:
> Did this problem recur in the latest 1.8.7 release

Not that I'm aware of.  If it's not working for you then please provide a full bug report.