micropython/micropython #2072: Adjustable verbosity for print output on REPL

torwag, 2016-05-13:
Hi, 
I am not sure whether this is a ESP8266 specific problem or not.
If you connect via serial to an ESP8266 board, depending on your current settings, you will see a lot of ongoing print messages in the terminal, guess the same is true for WebREPL.
E.g. I configured the ESP to connect to my wifi at home and using it at work I get every second:
```
no HomeNet found, reconnect after 1s
reconnect
f r0, scandone
```
It does not disturb the interactive code input, but makes it very tricky to write code.
E.g. try to write a functions having dozens of those messages in between. You literally have to type them blind.

I would like to suggest a general verbosity setting for the REPL.  A variable which users can set 
E.g. 
sys.verbosity=0 will print no outputs
sys.verbosity=3 will print very verbose ongoing outputs to the REPL.

It might be easily implementable by overloading the print function with a verbosity argument. This would allow to implement it in one go and adapt necessary print functions by time, without the need to refactor the current code-base. Furthermore, it would not trouble users normal usage of the print command, albeit user can choice to add verbosity settings to their own programs for debugging purpose.

A very first draft of the idea:

```
import sys

sys.verbosity = 1

def print(*args, **kwargs):
    if 'verbosity' in kwargs:
        if kwargs['verbosity'] <= sys.verbosity: 
            kwargs.pop('verbosity')
            __builtin__.print(*args,**kwargs)
    else:
        __builtin__.print(*args,**kwargs)
```

This would result in:
```
for sys.verbosity in range(0,4):
    print("Error-Level: {}".format(sys.verbosity))
    print('This is a error message about Pin {} and it tells you "{}!"'.format('1', 'I''m blown'),verbosity=2)
```
=== Error-Level: 0
=== Error-Level: 1
=== Error-Level: 2
=== This is a error message about Pin 1 and it tells you "Im blown!"
=== Error-Level: 3
=== This is a error message about Pin 1 and it tells you "Im blown!"

whereas the builtin print command will still work as normal
`print("I'm a normal print command")  `
=== I'm a normal print command

dpgeorge, 2016-05-13:
This is the same problem as #2005.

To disable these system error messages (they are not uPy errors/logs), use: esp.osdebug(None).

closed 2016-05-13 by dpgeorge

torwag, 2016-05-13:
Hi Damien,

thanks that solves the issue. Ah to bad I thought I had a nice idea about a general setting of the debug-level verbosity. But checking the existing python code it looks like there is not to much debugging options yet, which justify such an general implementation .

dpgeorge, 2016-05-13:
> thanks that solves the issue

Great!

I think the intention was to eventually disable OS-debug output by default, so you don't need to do esp.osdebug(None).

torwag, 2016-05-20:
Yep, I guess it would be better to make esp.osdebug(None) default.