micropython/micropython #2073: esp8266: I2C.writeto no longer accepts stop=False

deshipu, 2016-05-13:
The I²C functions used to accept a keyword ``stop`` argument, so that we could send a "repeated start" to the devices by performing several writes and/or reads with that set. However, it seems that the ESP8266 port no longer allows that argument:

```
>>> i2c.writeto(0x40, [1], stop=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function does not take keyword arguments
>>> i2c.writeto(0x40, [1], False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function takes 3 positional arguments but 4 were given
```

Is there a new way of achieving repeated start?

dpgeorge, 2016-05-13:
The "stop" argument was dropped, in favour of more primitive I2C operations, namely: start(), stop(), readinto() and write().  These allow you to construct arbitrary I2C commands.  Please see the docs: http://docs.micropython.org/en/latest/esp8266/library/machine.I2C.html?highlight=i2c#primitive-i2c-operations

deshipu, 2016-05-13:
Thank you for pointing this out. I made a pull request removing this from the quickref, then.

dpgeorge, 2016-05-13:
If you have any feedback on the use of these primitive operations (eg are you able to do what you want with them? can you do it easily and cleanly?) please let us know, thanks!

deshipu, 2016-05-13:
I'm trying, but I have a weird issue. The code I have initially was:

```
    def _get_reg8(self, address):
        data = ustruct.pack('<H', address)
        self.i2c.writeto(self._address, data, stop=False)
        return self.i2c.readfrom(self._address, 1)[0]
```

With the new primitives, it looks now like this:

```
    def _get_reg8(self, address):
        data = ustruct.pack('<cH', self._address << 1, address)
        self.i2c.start()
        self.i2c.write(data)
        self.i2c.start()
        data = ustruct.pack('<c', self._address << 1 | 0x01)
        self.i2c.write(data)
        data = bytearray(1)
        self.i2c.readinto(data)
        self.i2c.stop()
        return data[0]
```

However, when I try to run that, it gets ``OSError: I2C bus error`` as soon as it gets to ``i2c.write()``...

dpgeorge, 2016-05-13:
It looks like you are doing memory reads with a 16-bit address.  It should work.  The code you have looks the same as how the 8-bit memory reads are coded (see readfrom_mem).

You could probably make it simpler:
```
i2c.start()
i2c.write(ustruct.pack('<cH', self._address << 1, address))
i2c.readfrom(self._address, 1)
```

deshipu, 2016-05-13:
Found my errors, it was the weirdness of ``ustruct`` and its habit of silently ignoring errors. Got it to work, thanks!

deshipu, 2016-05-13:
I wonder if it would make sense to have a ``i2c.address(address, direction)`` or maybe optional ``address`` and ``direction`` parameters to ``i2c.start()`` to avoid some of the boilerplate.

deshipu, 2016-05-13:
The example above could be something like:

```
i2c.start(self._address, read=False)
i2c.write(ustruct.pack('<H', address))
i2c.readfrom(self._address, 1)
```

pfalcon, 2016-05-13:
> Found my errors, it was the weirdness of ustruct and its habit of silently ignoring errors. Got it to work, thanks!

Feel free to elaborate on that.

pfalcon, 2016-05-14:
>  or maybe optional address and direction parameters to i2c.start() to avoid some of the boilerplate.

As @dpgeorge explained, these are elementary I2C operations form which you can compose arbitrary I2C transaction. There're no concept of address at that level. Address is what you send after the start, it can be 8, 10 (actually 16), 0 or 1024 bits (up to you and your devices). 

> I wonder if it would make sense to have a i2c.address(address, direction)

So, you kind of start with presupposition that stop removed form writeto() is ultimate truth, and now try exhaustive search to find *workaround* somewhere else. You tried the lowest level where it does not belong, now trying random mid-level. If you know the whole picture (and I guess you do, if you're working with it), finding the right level to apply *solution* (not a workaround) is possible. So, if @dpgeorge is right and you're trying to do the same thing as .writeto_mem()/.readfrom_mem() do, except with 2-byte memory address, then that issue was already raised by others, it's that critical mass being talked about to find a right solution, and that's what should be done.
 
At the very least, there can be argumentation for restoring stop= for writeto (but it should be good, as usual, considering global picture, not just adhoc usecase).

deshipu, 2016-05-14:
@pfalcon I'm sorry, but I don't quite understand what you are talking about. Are we talking about me here, or about the I2C interface?

A "repeated start" condition is very much part of the I2C protocol, one of the 3 basic ways of using I2C. To do a "repeated start", you *always* send a start, followed by the device's address. This is very common, not only in the *_mem functions. I think it deserves a place.

When I see code like:

```
i2c.start()
i2c.write(ustruct.pack('<cH', self._address << 1, address))
i2c.readfrom(self._address, 1)
```

my skin crawls. Note how you have to use 8-bit address in one place, but 7-bit address in another. Note how you are mixing the protocol-specific bits with the data you are sending. And note how it is practically impossible, without good knowledge of the I2C protocol, to tell at first glance what is going on there.

deshipu, 2016-05-14:
>>Found my errors, it was the weirdness of ustruct and its habit of silently ignoring errors. Got it to work, hanks!
> Feel free to elaborate on that.

Well, I submitted a bug report report about that a while ago, but it was closed and I kinda agree with the reason why it was closed, so that's something we have to live with. The problem came from the combination of it being 2AM and me not quite remembering which formatting characters ``ustruct`` supports compared to standard Python ``struct`` (I vaguely remembered b/B were not supported, but seems like they are now), compounded with the fact that ``ustruct`` silently ignores errors such as unknown formatting characters or wrong number of parameters passed. The effect is the above:

```
i2c.write(ustruct.pack('<cH', self._address << 1, address))
```

which doesn't *actually* work, and should *really* be:

```
i2c.write(ustruct.pack('<BH', self._address << 1, address))
```

But of course, with limited resources we can't have everything, and if a little less convenient use is the price I have to pay for conserved resources, so be it.

pfalcon, 2016-05-14:
> Are we talking about me here, or about the I2C interface?

I'm talking about you, a long-time lurker, and now a regular contributor, giving an example of counter-productive handling of issues (looking for random places to apply workaround) vs trying to find right place to apply fix - in the hope that it will let you use your time more productively.

> This is very common

Prove that. https://github.com/micropython/micropython/issues/1430 is good place to contine teh discussion. Then prove that it's possible to do that with HW I2C on all MCUs/SoC.

> Note how you have to use 8-bit address in one place, but 7-bit address in another.

API Levels. I have told about that. If you choose to mix different API levels, it looks weird, but it's your choice (and sometime there may be even reasons to do that). There're currently 3 levels. The lowest (.start, etc.) is complete, there's nothing to take or add to it. The highest lacks support for different intra-device register address widths. Normally, the lowest and highest standard API would be enough for project like MicroPython, yet we have that middle-level, which is neither as powerful as lowest, nor as convenient in common cases. Prove that it needs further elaboration and not complete removal to stop confusion.

deshipu, 2016-05-14:
> I'm talking about you, a long-time lurker, and now a regular contributor, giving an example of counter-productive handling of issues (looking for random places to apply workaround) vs trying to find [...]

I see. I'm not interested in discussion about me, or about proving anything, so I'm closing this issue. Let me know when you want to have a technical discussion.

pfalcon, 2016-05-14:
I'm always having technical discussions. Please don't close this issue (just unsubscribe), it will wait to collect critical mass of thoughts from people interested to resolve it.

pfalcon, 2016-05-14:
> Well, I submitted a bug report report about that a while ago

uPy evolves all the time, some issues aren't resolved immediately, but require some critical mass gathering and/or clarification of direction (for example, the question of sticking to minimal possible code size becoming a bit less relevant with flash size growing, while user friendliness becomes more important with uPy getting more and more popular). This issue was a TODO in the code and now fixed: https://github.com/micropython/micropython/commit/e53fb1bf031a53ad23a45846addf14e25cebb97d

pfalcon, 2016-05-14:
> This issue was a TODO in the code and now fixed: e53fb1b

ca63c770733ead7fd5de061ddc5268ffbd3c368b also documents which type codes are supported.

deshipu, 2016-06-22:
By the way, nodemcu has ``i2c.address()``: http://nodemcu.readthedocs.io/en/master/en/modules/i2c/

dpgeorge, 2016-06-28:
> By the way, nodemcu has i2c.address()

They seem to follow the Arduino API (cf beginTransmission).

deshipu, 2016-06-28:
>> By the way, nodemcu has i2c.address()

> They seem to follow the Arduino API (cf beginTransmission).

Arduino's ``Wire.h`` library puts the address in ``beginTransmission``, though, (see https://www.arduino.cc/en/Reference/WireBeginTransmission). Nodemcu's API looks more like Micropython's "low level" API (they also have basic ``read()`` instead of separate ``requestFrom``, ``available`` and ``read``). Here's a quick comparison table:

```
+----------------------+----------------+---------------+---------------------+
| Function/Environment | Microptyhon    | Nodemcu       | Arduino             |
+----------------------+----------------+---------------+---------------------+
| Initialization       | i2c.init()     | i2c.setup()   | begin()             |
| Start Condition      | i2c.start()    | i2c.start()   | N/A                 |
| Send Address         | N/A            | i2c.address() | N/A                 |
| Start Cond.& Address | N/A            | N/A           | beginTransmission() |
| Write Data           | i2c.write()    | i2c.write()   | write()             |
| Request Data         | N/A            | N/A           | requestFrom()       |
| Check Availability   | N/A            | N/A           | available()         |
| Read Data            | i2c.readinto() | i2c.read()    | read()              |
| Stop Consition       | i2c.stop()     | i2c.stop()    | endTransmission()   |
| Receive Callback     | N/A            | N/A           | onReceive()         |
| Request Callback     | N/A            | N/A           | onRequest()         |
+----------------------+----------------+---------------+---------------------+
```

deshipu, 2016-10-24:
In addition, from the specification at http://www.nxp.com/documents/user_manual/UM10204.pdf on page 14:

> 4. I 2 C-bus compatible devices must reset their bus logic on receipt of a START or
> repeated START condition such that they all anticipate the sending of a slave
> address, even if these START conditions are not positioned according to the proper
> format.
> 5. A START condition immediately followed by a STOP condition (void message) is an
> illegal format. Many devices however are designed to operate properly under this
> condition.

dpgeorge, 2016-12-02:
I2C methods readfrom and writeto now accept "stop" as the last parameter to the methods.  See 37333cb00a9e08961045b444a4b090c358e57b1c.

closed 2016-12-02 by dpgeorge