micropython/micropython #2075: Embedding micropython as a library (on unix)

hauptmech, 2016-05-14:
I guess this is a continuation of #1335 and #1279. Another data point for the discussion.

I was looking for a lightweight python implementation to embed. I found micropython, and after playing with it a bit decided to look at embedding it. 

I started by reading unix/main.c. Not too complicated, but tightly integrated. The global context variable makes me nervous.

I read (but did not try) the two patches above. In spite of the discussion about complexity around #1279, I feel more comfortable with it because it creates the feel of a library, which is exactly what I want.

 If I can grab the source, build a static and/or dynamic lib, and shove it in a system directory with an accompanying pkg-config file I'm happy.  I think the worries about distributions and packaging are unfounded; I regularly create packages for Archlinux and Debian-ish systems when official ones are not available. I think that if you generate the standard outputs (lib files, include file, canonical executable), packagers will take care of the rest. 

Having to locate and path-to and build the micropython source when I'm building my own source, as #1335 implies I should, feels a lot more risky.

@pfalcon Mentioned that the code is the documentation, and I totally agree as long as there is a nice comment at the top outlining the big picture. 

If micropython on unix built as a library and main.c included the key include file and linked against the library, I'd need nothing more.

Anyway, sorry I don't have time to offer a patch or something. Hopefully the additional opinion is helpful.

pfalcon, 2016-05-14:
@dpgeorge: Are you now ready to merge my embedding example from #1335? Ignoring this area doesn't make sense, people are asking about it regularly and not having even basic support for that undermines MicroPython comparing to other projects. And my patch is so far the cleanest initial step.

hauptmech, 2016-05-15:
I had a second read through #1335 and #1279 to try to understand what's happened so far and give a clear opinion.

In short, my desire is for micropython in library form. Lua is the current tool I'm using and is probably a good benchmark for what embedding a scripting language via library should look like. 

It seems that @pfalcon sees his patch as a step towards that, but I'm not seeing his logic. If the current unix main used micropython as a library, it would be fine. It is not too complex. 

A quick note a @pfalcon's proposed example:

> So, an example I had in mind for this is: a C app reads file line by line, and writes out each line. But it allows user to specify a python file, that file should contain definition of function "process", a C app will run each line thru that function and write out returned result.

This would not be a useful example for me as it is no different than the REPL. Calling into python functions from C and extending the python with C functions would be more useful for me.

As a side note, I agree with @stinos that a single init function with the most robust GC configuration being set up inside is desirable.  Even though it's wrapping very simple logic, it saves the new user from searching for GC documentation / code and trying to make a decision about what a good configuration should be.

@dpgeorge's python script to inject the context as a function parameter sounds like the best option in spite of the fact that "it's not ready". It sounds like it's doing the right thing to make it a library.

However, a library that uses global variables and only allows a single instance of micropython to be used would be ok too. A little bit scarier, but if it were easy to verify which variables were global, one could probably convince themselves that it would be ok to use. 

I have a short attention span but if you guys decide a clear direction to move in regarding a micropython library I'm happy to help move things forward.

stinos, 2016-05-15:
> In short, my desire is for micropython in library form

To play around with that basically all you have to do is built a static library instead of a binay, after having removed STATIC from the functions you're going to need (execute_from_lexer etc, mostly in main.c). To make that reusable you'll inevitably end up with something like #1279. Don't worry too much about the global state: in practice it is not a problem as long as you stick to 1 uPy instance (well, you have to) and only call into uPy from one and always the same thread.

I agree with you that #1335 doesn't immediately look like a step towards embedding (depending on the interpretation of that term), I see it more as an example of the minimum which has to be copy/pasted from e.g. unix' main.c to get uPy code interpreted and ran. It does work, but I'm not sure it's worth adding it as such for pretty much the reasons dpgeorge mentions in the comments. And also because exactly the same can be achieved by e.g. a paragraph in the docs saying 'if you want a minimal embedding sample, take minimal's main.c and replace the REPL loop with ``do_str("print('Hello world of easy embedding!')")``'. 

Anyway, in the end I still think the ultimate way to go is dpgeorge's idea of generating an amalgamation from the source, all functions prefixed with micropython_ and global state replaced with context. Also because from there you can go anywhere. Want a static library? Just generate the source and build it into a lib. Want to embed in your own project? Just add the single uPy source file.

hauptmech, 2016-05-15:
@stinos 
> Anyway, in the end I still think the ultimate way to go is dpgeorge's idea of generating an amalgamation from the source, all functions prefixed with micropython_ and global state replaced with [...]

I agree as long as these are build/make options. It's ok if there are preprocessors on the front end, but if one has to start reading the code to understand what to do, it's too high a barrier to entry. 

I did a read through of all the code (/py /unix) today and after that I feel I understand mostly where the boundaries between the porting interface fuctions and the useless-to-me REPL main.c stuff. However, I'm not sure I can use micropython even though I want to because I can't tell users of my stuff to just go pull the source and build it because it does not work that way. Instead I have to include a copy of the source (or a submodule) with my own source. Definitely not preferred.

hauptmech, 2016-05-15:
To help myself prepare for using what's in /unix now, I did my own version of a minimum first step in making a library. All I did was move what I understood to be porting specific stuff to it's own file,` unixport.[c,h]`. The library build includes everything except for main.c.

The changes to the code are minimal; removing the option for `compile_only`, making `emit_opt` a function argument, and not using the `mp_printf` for the generic errors in `main()`. Everything else is just cut-n-paste.

`make lib` "slurped" 408b74d74ce8de806f8718a02d350b8326d89361 too much so I hacked the mkrules.mk and Makefile to build. I'm sure there is a better way.

One could argue that `handle_uncaught_exception`, `execute_from_lexer`, `do_file`, and `do_str` belong in main.c for this split. I would not disagree, the decision I made was arbitrary.

One comment I'd like to make is that for myself, between having a working library build and having the nice initialization, execute_from_lexer, and repl helper code, the library build is more important. I'm going to be learning the API anyway and getting some practice by adapting what's in main.c to my own purposes is fine. However, figuring out where exactly the split in the code for the library should be, and what modifications to the Makefiles will do to this multiarchitecture beast is a lot scarier.

dpgeorge, 2016-05-16:
> Are you now ready to merge my embedding example from #1335?

Nope.  I still stand by my comments regarding that PR.

pfalcon, 2016-05-16:
> Nope. I still stand by my comments regarding that PR.

That's very-very said. Why it was proposed is to give the initial right direction and avoid a case when people in the name of "embedding" munge with completely unrelated code (like unix port). That's what original @stinos' patch did, and the story repeats 1:1 here again! Not having a good starting point causes people to go in loops. And just as the lack of persistent bytecode support (now alleviated) hurts MicroPython image and opportunities, just the same the complete lack of embedding support hurts it too (damn, I can't even suggest uPy as a candidate for embedding, because there's literally nothing what people can start with). Comments you gave in that PR are technical specifics and future points of work, which can be resolved going forward, but there's simply nothing to start with now, and my patch at least shows that simple embedding can be achieved right away, and how to do it without flying to Mars. You're essentially blocking any progress of uPy in that direction for an unknown reason.

dpgeorge, 2016-05-16:
> Why it was proposed is to give the initial right direction and avoid a case when people in the name of "embedding" munge with completely unrelated code (like unix port)

But it doesn't give the right direction.  It's just another simple example how to make a stand alone executable.  This is already provided by minimal/.  And being part of the unix/ directory just adds to confusion and complexity (especially the unwieldy makefile, see #1977).

hauptmech, 2016-05-16:
@dpgeorge Do you have an idea (or better, a clear idea) of how you would like to see the desktop ports and library and interpreter organized?

pfalcon, 2016-05-16:
> But it doesn't give the right direction.

Of all currently available examples, it's the most suitable by the formal criteria of not changing anything not related to the embedding feature.

> It's just another simple example how to make a stand alone executable.

No, it **embeds** MicroPython in a `int main() {}` C application. Of course, the end example is "standalone executable", what else would you expect?

> This is already provided by minimal/.

This is a logical mistake similar to https://github.com/micropython/micropython/pull/2080#issuecomment-219312902 . The fact that in particular point in space and time they look similar at some (low) level doesn't mean they're the same concept or notion. They different purpose and each will develop in its own direction.

> And being part of the unix/ directory just adds to confusion and complexity (especially the unwieldy makefile, see #1977).

I proposed to move it to examples/embed/ . If you think we don't have enough top-level dirs and need more, it can be that. Please specify what needs to be changed to have that *minimal* code to showcase embedding merged.

stinos, 2016-05-17:
> avoid a case when people in the name of "embedding" munge with completely unrelated code (like unix port)

The code being munged cannot be truly unrelated else it wouldn't have to be touched. Your embedding sample does it as well actually: sure, you didn't modify any unix port code, but you basically did copy/paste it. To go anywhere near embedding one's always going to need execute_from_lexer and similar functions. So as @dpgeaorge says: sticking them in a directory makes a lot of sense to me - most of the stuff in unix main is after all not port specific. In fact if that's done, your minimal embedding sample would become even more minimal (like, reduce code size by almost factor 2) since you could just reuse execute_from_lexer. But even then, the sample would still exactly the same as the minimal port's main but with the REPL replaced by a single do_str call.

pfalcon, 2016-05-17:
> The code being munged cannot be truly unrelated

You're talking about code, I'm about process. The process of adding embedding support should not involve changes to existing code, as long as it's possible to avoid these changes. Not following this simple idea leads to mistakes, like removing random (not related to embedding) features from existing code, or just hard to prove or review changes.

> but you basically did copy/paste

So what? That's how you *start* doing embedding. Any other start is worse.

> To go anywhere near embedding one's always going to need execute_from_lexer and similar functions. 

Yes, as one of the next steps! So, we understand each other - complex tasks are broken down into small and easy steps (easily reviewable in our case), and they are executed in order, starting from the start. That was said a year ago, https://github.com/micropython/micropython/pull/1335#issuecomment-112961631 , and as it's simple concept, it was as understandable then as it is now.

> In fact if that's done ...

That phrase would better be spelled as "In fact after it's done ...", and the keyword of the whole sentence is **after**. Small steps done one after another, remember? And what's interesting, it won't be directly related to "embedding" at all. It will be just application of a general refactoring principle "If A and B duplicate the same code, it should be factored out for reuse".

> But even then, the sample would still exactly the same as the minimal port's main but with the REPL replaced by a single do_str call.

OMG, that's something. All things in the world consists of the same atoms in the same ways. Does it make all things in the world the same? A: No. "minimal port" and "embedding example" are different concepts no matter the fact that they consist from chars of the same alphabet or even in the same configurations. Telling someone that they should use "minimal port" as an "embedding example" is wrong, and will make a receiver go into wrong loops with the poor minimal port.

hauptmech, 2016-05-18:
What I would want from micropython is:

1. A canonical port to unix (no main function, whatever top level API you want to provide, I'll re-port if I don't like how you have linked into the OS).
2. A canonical application. stdin -> stdout & stderr. REPL nice to have but optional.

The canonical application serves just fine as an embedding example. (Further documentation and tips & tricks is always welcome.)

The whole point is to draw a clear line between micropython and my application that I can trust. To have an interface I can understand without reading all the code.

I keep reading the past issues trying to understand what a good way forward is. Between @pfalcon being too conservative (the goal is not that clear either) and @dpgeorge being too radical, and neither approach would have helped me. You guys need to find a middle ground. 

@stinos's patch would have let me start playing with micropython in my own application without having to read all the code. I happen to like reading compiler guts so I would have anyway, but I bet you are missing a lot of potential adopters (future project participants) on the unix side.

Currently Lua is the defacto standard for embedding scripting into an application; with the other options having very small code bases. A lightweight python implementation that can leverage the python code base and compete with Lua on size and ease of embedding is a very attractive idea.

Meanwhile, I'm going to see if I can use `libmicropython.a` as is. The linker does not seem to get confused by the main symbol, I'll have to see what symbols I need are not available.

stinos, 2016-05-18:
@pfalcon so let me get this straight: your proposal is to merge your (or similar) embedding sample, and then as next steps something like (not in particular order):
- provide a common implementation of e.g. do_str/execute_from_lexer and use it in unix main, embedding sample, ... (and this will likely split in multiple commits). This implementation would then be the toplevel API to 'run' uPy. (basically parts of my PR)
- add way to build a library (basically this PR)
- implement embedding sample in terms of uPy library, using seperate Makefile?
- ... (not sure what comes next, maybe it's done by then)

If so I can see why you want it that way, you could sort of follow the progress made on the embedding sample, though on the other hand I'd be fine with the embedding sample only being added in step 3

pfalcon, 2016-05-18:
>  your proposal is to merge your (or similar) embedding sample, and then as next steps something like (not in particular order):

Yes, exactly. Next steps will be making changes from your and @hauptmechp's patches, but very preferrably in small, easy (== fast) to review steps (like function by function).

> If so I can see why you want it that way

Please remember, that with the original discussion of the embedding a year ago there was a parallel discussion "why people prefer to submit non-mergeable patches, even though it's counter-productive for everyone" (ones which are hard to review and pinpoint issues, so they wait in queue months, soon years). Doing it all in small, easily provable steps will allow for much faster progress then trying to lump it up.

> add way to build a library (basically this PR)

There's a way to build library for a long time (make lib).

pfalcon, 2016-05-18:
> Meanwhile, I'm going to see if I can use libmicropython.a as is.

@hauptmech : That's very reasonable approach, one which each of us have to use when some software doesn't work exactly as we'd like/expect. And thanks for continuing to give it a try, while we're working out details of how to bootstrap "official" support for embedding. Rest assured that my personal aim (one of) is to make MicroPython a viable choice for usecases where Lua was almost the only option before, and we're ready to take any good traits from it, but not blindly. And our default affinity is still (C)Python, and that's still after trying to find what's the best way to do it in a way which fits MicroPython's aim of minimalism well.

pfalcon, 2016-05-18:
> Meanwhile, I'm going to see if I can use libmicropython.a as is.

FYI, I used it for Android port, and yep, it worked. So, embedding is supported, it's just "hidden" and requires digging in. The whole issue is how to expose it in evolutionary, not revolutionary way.

pfalcon, 2016-05-19:
> Please specify what needs to be changed to have that minimal code to showcase embedding merged.

@dpgeorge : It would be nice to get some response. I'm asking now because it's Thurs and I can do any reasonable work only on weekend, and you usually not around then, so mis-syncs may put it off for week(s). (And yep, it may be not the best time, but as you said yourself, it would be nice to tackle older issues which drag down, and this simple (from my PoV) thing waits in queue for a year already). Thanks.

hauptmech, 2016-05-20:
@dpgeorge

dpgeorge, 2016-05-28:
Please see #2122.  Sorry I don't have time for an extended discussion on this.

closed 2016-05-28 by hauptmech

psprint, 2021-05-15:
What is resolution of this issue? Can be a µ-python interpreter easily embedded inside a C program?