micropython/micropython #2076: Work out how to run file tests on boards

pfalcon, 2016-05-14:
Currently, file tests don't run on boards, because "io" subdir is completely skipped. Proposal: split it in 2, io-local and io. io-local may depend on the pre-availability of test data files locally (and will be skipped for boards). Tests in io will create any needed test files or test other aspects altogether (e.g. io.StringIO).

dpgeorge, 2016-05-16:
Sounds okay.  Issues are 1) running io test regularly on a board can wear out the flash, 2) board may have existing files that interfere with tests.  But I guess one uses a dedicated test board for the tests and would expect to deal with these (small) issues.

pfalcon, 2016-05-16:
> 1) running io test regularly on a board can wear out the flash

If that's really your concern, we can either add "exclude" switch or still not include it by default, just have such tests in their own dir for easy running.

pfalcon, 2016-05-16:
Well, we'd need 3 dirs - one for non-FS io.* stuff (maybe move to some other dir), 2nd for stuff which requires local files pre-present, and 3rd for self-contained io tests.

dpgeorge, 2016-05-17:
3 separate dirs is ok, maybe io_generic/ for non-FS stuff?  Or even just put these in basics/ with filename prefixed by `io_` (sys tests are in basics and modio is in py/ just like modsys).

Names for other dirs can be io_file/ and io_localfile/?  Whatever, but please use underscore for consistency with feature_check dir.

pfalcon, 2016-05-17:
> Or even just put these in basics/ with filename prefixed by io_ (sys tests are in basics and modio is in py/ just like modsys).

I'd prefer this too, as creating extra dir for 2-3 files with little potential for big growth doesn't make sense.

> Names for other dirs can be io_file/ and io_localfile/?

My idea was to just keep existing "io" dir for self-contained tests, but if you prefer longer names like that, ok.

> Whatever, but please use underscore for consistency with feature_check dir.

Ok!

dpgeorge, 2017-06-14:
With the addition of VFS, an alternative to the above options is to create the files on the fly in a RAM disk.

dpgeorge, 2022-06-23:
Closing due to inactivity.

closed 2022-06-23 by dpgeorge