micropython/micropython #2085: Getting ImportError: no module named 'pyb'

tscrip, 2016-05-17:
I am getting **ImportError: no module named 'pyb'** whenever I try to run the command: **import pyb**. Anyone have a suggestion?

dpgeorge, 2016-05-17:
What board/platform are you using?  Note that the ESP8266 does not have a pyb module.  Use the machine module instead (see the docs and tutorial).

tscrip, 2016-05-17:
It is the ESP8266. All of the documentation I have looked at says to use pyb. I will try the machine module.

dpgeorge, 2016-05-17:
> All of the documentation I have looked at says to use pyb.

Please can you tell what this documentation is?  It should be updated.

mcauser, 2016-05-18:
@dpgeorge are you planning on renaming pyboards pyb module to machine in the future?

pfalcon, 2016-05-18:
@mcauser : No, pyboard has own machine module with now standard across ports interface. Help with filling it in with more stuff (ported from pyb) is welcome.

pfalcon, 2016-05-26:
Assuming issue resolved.

closed 2016-05-26 by pfalcon

JJaniel, 2018-10-18:
> I am getting **ImportError: no module named 'pyb'** whenever I try to run the command: **import pyb**. Anyone have a suggestion?

Hi, 
This link will Help You, pyb module (found in pyboards) has been changed to machine module ( for 8266 node MCU) and here below link shows how the code gets changed.....It will solve your issue...Good Day
https://primalcortex.wordpress.com/2016/05/31/esp8266-and-the-micropython-firmware/

wardboumans, 2020-12-01:
Just ran into this issue. Very confusing since its the first example on the homepage that I copy pasted as a quick test on my Lolin32 board still uses 'pyb'.

robert-hh, 2020-12-01:
The pyb module only exists in the Pyboard ports. For ESP32 and new scripts please use the machine module.

wardboumans, 2020-12-01:
I know now, i'm just saying its confusing for MP starters.

cree8, 2020-12-02:
Is there an equivalent _machine_ library for the _servo_ library contained in _pyb_? The _servo_ library is a better way of controlling servos (timing) as opposed to manually controlling them via freq and pwm commands..