micropython/micropython #2098: esp8266: no outout from SPI

deshipu, 2016-05-22:
The ``machine.SPI`` interface doesn't seem to output anything on the gpio pins when ``spi.write()`` is called.

Test program:
```
>>> from machine import SPI, Pin
>>> spi = SPI(miso=Pin(12), mosi=Pin(13), sck=Pin(14))
>>> spi.write("abc")
```

Expected result: activity on the gpio pins 12, 13 and 14 corresponding to the data being sent.
Actual result: the pins all remain low.

Perhaps I'm doing something wrong, or forgot to call something first?

deshipu, 2016-05-22:
Initializing the pins to OUT mode before initializing SPI makes them work.

deshipu, 2016-05-22:
Currently to work, SPI has to be initialized like this:

```
>>> from machine import Pin, SPI
>>> spi = SPI(miso=Pin(12), mosi=Pin(13, Pin.OUT), sck=Pin(14, Pin.OUT))
```

Either we need to update the documentation and examples, or set the correct modes in the SPI's init.

dpgeorge, 2016-05-26:
This should be fixed by 7e809b4baf7fed0b88afa56a297f9c69d4d337ab.

closed 2016-05-26 by dpgeorge