micropython/micropython #2100: webrepl doesn't work if boot.py is running

mangelajo, 2016-05-22:
For example, with:

```
import webrepl
import time

webrepl.start()

while True:
    time.sleep(1)

```

It's possible to connect to webrepl, but you can't do anything (even uploading new files).

We should be able to break current execution, and to upload files.

deshipu, 2016-05-22:
I think you want to put that loop in ``main.py`` and not ``boot.py``.

mangelajo, 2016-05-23:
Ahaa!, 

What's the difference between main.py and boot.py in that regard, do we have any documentation for that?  I see that webrepl accepts connections, but nothing else happens.

I will have to try it, I need to disassemble my device (micropython in a coffe maker) and probably kill the execution of boot.py via CTRL+C if that's possible, or reflash.

Thank you.

deshipu, 2016-05-23:
> What's the difference between main.py and boot.py in that regard, do we have any documentation for that? I see that webrepl accepts connections, but nothing else happens.

I don't think there is a difference -- it's just semantics. The settings should go into ``boot.py``,  the program into ``main.py``.

I just tested, and it is indeed impossible to interrupt a running program from the webrepl. From the serial console you can press "ctrl+c" to interrupt it, but that doesn't work over the webrepl.

mangelajo, 2016-05-23:
CTRL+C does not work for me in the serial console (I'm using screen /dev/tty.... 115200),

btw I found that, when you connect via webrepl, and the main loop is running, you get this:

```
WebREPL connection from: ('192.168.1.7', 62452)
Fatal exception 29(StoreProhibitedCause):
epc1=0x401012a6, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

```

pfalcon, 2016-05-23:
I can confirm this behavior - if an application is running, connecting via webrepl will show application output, but won't accept input. Also, a running application may crash afterwards in Ctrl+C is pressed in *UART* REPL.

Yes, that's not how we intended it to be, the idea was that connecting via webrepl would indeed allow to stop a running app for example. It's hard to say what's wrong - this looks like a race condition somewhere or maybe even crash due to stack overflow. We'll be investigating it and optimizing webrepl going forward, unfortunately now it's another known issue.

pfalcon, 2016-05-23:
Everything is better on docs front - the boot process in particular is documented in http://docs.micropython.org/en/latest/esp8266/esp8266/general.html#boot-process

dpgeorge, 2016-05-23:
Note that ctrl-C handling is done in UART rx interrupt handler.  There will need to be some corresponding special handling to check for ctrl-C in incoming webrepl data.

pfalcon, 2016-05-23:
Ok, so with @dpgeorge's hint this issue fixed in master and works well in my testing. @mangelajo , please test on your side.

mangelajo, 2016-05-24:
ack @pfalcon , I will test, do we have any autobuild of master somewhere?, or should I proceed with vagrant?

Notes: I looked at the documentation searching for boot.py and main.py, and search didn't work: http://docs.micropython.org/en/latest/esp8266/search.html?q=main.py&check_keywords=yes&area=default

And I also looked at the commit history since 1.8, and didn't see anything related to webrepl, but probably I missed it if it's fixed in master, I will look again.

mangelajo, 2016-05-24:
ahh https://github.com/micropython/micropython/commit/4681b86850076c013be2477d3850fd536361b62e

Thank you, that was quick,

pfalcon, 2016-05-26:
@mangelajo : So, I assume the issue was fixed for you, closing.

closed 2016-05-26 by pfalcon

mangelajo, 2016-06-03:
Well, I didn't manage to get a build of the firmware. So I can't tell, I'll check eventually on next release.

It'd be great to get an autobuild of the firmware for master commits.

mangelajo, 2016-06-03:
Thank you for fixing this.