micropython/micropython #2103: Help!   HSI of STM32 is not supported!

1983FFly, 2016-05-23:
In file:System_stm32.c
    line:272

[image]
[image]

1983FFly, 2016-05-23:
Change function SystemClock_Config(void) of system_stm32.c

void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct;
  RCC_ClkInitTypeDef RCC_ClkInitStruct;

  __PWR_CLK_ENABLE();

  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);

  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSICalibrationValue = 16;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSI;
  RCC_OscInitStruct.PLL.PLLM = 16;
  RCC_OscInitStruct.PLL.PLLN = 336;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
  RCC_OscInitStruct.PLL.PLLQ = 7;
  HAL_RCC_OscConfig(&RCC_OscInitStruct);

  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV4;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV2;
  HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_5);

  HAL_SYSTICK_Config(HAL_RCC_GetHCLKFreq()/1000);

  HAL_SYSTICK_CLKSourceConfig(SYSTICK_CLKSOURCE_HCLK);

  /* SysTick_IRQn interrupt configuration */
  HAL_NVIC_SetPriority(SysTick_IRQn, 0, 0);
}

void HAL_MspInit(void) {
#if defined(MCU_SERIES_F7)
    /* Enable I-Cache */
    SCB_EnableICache();

    /* Enable D-Cache */
    SCB_EnableDCache();
#endif
}

1983FFly, 2016-05-23:
[image]
[image]

chrismas9, 2016-05-24:
USB is not supported on the F4 using HSI. USB requires +/-0.25% accurate clock and HSI is only 1 - 4% depending on temperature. That's probably why MicroPython doesn't support HSI.

You could use the accelerometer pins PB2 and PB5 for LEDs and add a crystal. PB2 LED would need to be tied to GND. Since your board is so tiny (well done!) you could also use a MEMs oscillator as they are much smaller than crystals. 12MHz crystals are smaller than 8MHz crystals.

dpgeorge, 2016-05-24:
If you don't care about USB working then you can use the HSI by calling machine.freq(16000000).  But this setting won't persist across resets, and you can't configure it statically / at build time.

1983FFly, 2016-05-24:
Thank you.I will use a MEMs oscillator!

dpgeorge, 2017-06-14:
As discussed above HSI is only partially supported because USB relies on an accurate clock.

closed 2017-06-14 by dpgeorge