micropython/micropython #2104: Frozen bytecode compilation fails on strings containing some quotes

ryannathans, 2016-05-23:
I have a file called that contains the following

```
foo = "\"bar\""
```

and another that contains the following (single quote, double quote, double quote, single quote)

```
foo = '"bar"'
```

both throw errors during frozen bytecode compilation, it seems that the parser treats the quotes incorrectly.

```
build-STM32F439/scripts/frozen_mpy.c:1320:39: error: expected '}' before 'bar'
         (const byte*)"\xd4\xe5\x05" ""bar"",
```

It's strange, because the following works (double quote, single quote, single quote, double quote)

```
foo = "'bar'"
```

ryannathans, 2016-05-23:
foo = """\"bar\""""

also fails compilation. it seems as though if any string contains " in the final interpreted string, the frozen compilation breaks

dpgeorge, 2016-05-23:
Should be fixed by 202d5acd06d320f3d2485b85d4ff6c412e50ad21.

ryannathans, 2016-05-27:
Confirmed, fixed.

closed 2016-05-27 by ryannathans