micropython/micropython #2105: Restore PDF manual generation

pfalcon, 2016-05-23:
When hosted on readthedocs.org, there was an option to download a PDF manual. Nowadays, we can't even brag how many pages it has. Would be nice to get them generated.

mcauser, 2016-05-27:
This? http://media.readthedocs.org/pdf/micropython/latest/micropython.pdf

pfalcon, 2016-05-28:
@mcauser : You nailed the problem right. @dpgeorge would need make the changes to the docs.micropython.org to publish variants for all supported ports.

pfalcon, 2016-06-05:
@dpgeorge : If you have any free cycles, I guess this ticket is worth looking into. Because it's the easiest way (we have on our hands) to address tickets like https://github.com/micropython/micropython/issues/2126 - by telling people "if in doubt, search PDF".

dpgeorge, 2016-06-06:
To my surprise I was able to install latex et al on the micropython.org server without breaking the wiki (ie Ruby).  Now PDF generation is enabled and linked from the sidebar, and should be updated along with the HTML.

profra, 2016-06-06:
It seems that a lot of info for WiPy is inside of the generated PDF... 
but every reader assumes reading info for esp8266 only ...

pfalcon, 2016-06-06:
@dpgeorge : Cool, thanks! But something's not right - sections of docs are missing. Locally generated (`make MICROPY_PORT=esp8266 latexpdf`) PDF for esp8266 is 120 pages. http://docs.micropython.org/en/latest/micropython-esp8266.pdf is however only 66 pages. As an example, all the subsections of "2.1 Python standard libraries and micro-libraries" are missing.

dpgeorge, 2016-06-06:
On my local machine (very similar to micropython.org server), `make MICROPY_PORT=esp8266 latexpdf` gives a 72 page document, with 2.1 subsections missing.  Don't know why.

pfalcon, 2016-06-06:
@dpgeorge : I'll try clean build when get a chance. And if you have a chance, please provide a log of your run of "make MICROPY_PORT=esp8266 latexpdf".

pfalcon, 2016-06-06:
So far I just checked that even on a clean repo clone (vs my working copy) it builds 120-page doc on my system.

pfalcon, 2016-06-06:
Ok, so I have sphinx 1.2.2. Trying 1.4.2, I get the same broken behavior as @dpgeorge . Trying 1.4.3 (the latest), it's completely broken - cannot find some stylesheet, probably existing latex too old for it, lol.

dpgeorge, 2016-06-07:
Yes I use Sphinx 1.4.2.  Latex is pretty stable nowadays so it's strange that it would break like this.

pfalcon, 2016-06-07:
Well, it's not latex what breaks, but sphinx. I started looking into that, but actually ended up looking into how to resolve HTML index/search issues (picking up modules not included in the current docset). The solution is coming, in the form of patched sphinx. I spent all night on that, so don't think I'll be looking into it anytime soon. So, patched 1.2.2, in the form of git submodule, is the solution to all problems. (Well, my patches are against 1.4.2, so need to be backported first and verified for not regressing PDF generation).

pfalcon, 2016-06-07:
Ok, with my cute sphinx plugin, 1.2.2 starts to behave similar to 1.4.2 wrt to PDF generation - ~82 pages are generated, section 2.1 is empty. Love the tool!

pfalcon, 2016-06-07:
In 120-page doc, all content is duplicated: first time top-level, then everything under "MICROPYTHON DOCUMENTATION CONTENTS" title. Love the tool.

pfalcon, 2016-06-07:
I pushed Sphinx plugins mentioned above to https://github.com/pfalcon/sphinx_sane_only . Going to commit them to uPy repo and enable tomorrow.

pfalcon, 2016-06-09:
Ok, so duplicate content is due to the way TOC is organized: only few chapters are made top-level, and complete chapter list is under "MICROPYTHON DOCUMENTATION CONTENTS". That may make sense for HTML docs (hmm), but as can be seen, Sphinx 1.2.2 just renders everything in dup. And it's actually a fix in 1.4.2 that it avoids to render dup content.

Missing 2.1 section is due to fact that despite all Sphinx' authors efforts, pickled doctree produced for one output format doesn't render too well with another output format. So, to achieve bliss, build/*/doctrees/ should be removed before rendering each doc type.

The final ingredient are my extensions, which continue to elaborate, and with all the latest changes, the docs really shape up into something nice and clean. So, further steps:

1. Standardize of Sphinx 1.4.2.
1. Push https://github.com/pfalcon/sphinx_selective_exclude to our repo.
1. Make a custom TOC for PDF.

pfalcon, 2016-06-09:
And yeah, all this takes *a lot* of time, instead of doing "useful work". But without good docs (and we pretty OK on content, but not so good on structure and indexing), we can't even say people RTFM.

pfalcon, 2016-06-11:
Results of work above pushed in https://github.com/micropython/micropython/commit/f6d01b8b67325ec6c5e48251f042504448b57a7a and https://github.com/micropython/micropython/commit/91031b60dc974ab76159c9c2e2f0ef5c453f88ee . All issues mentioned above are also understood now: they're related to use of stale cached doctree (with eager_only plugin active, a cached doctree is builder specific) and peculiarities of ToC which was intended for HTML output.

With these changes and following simple rules above, docs generate without unexpected artifacts (like stuff which should not be there is included or mentioned, or stuff which should be there is missing). We'll see how it works on docs.micropython.org and how to deal with any issues there.

After that, preparing good TOC for PDF (and idea to have a single TOC is also viable IMHO) should be last task, with only adding more docs after that.

pfalcon, 2016-06-19:
So, it was clear next day that my changes above helped, by I only now got to review all 4 generated PDFs - they have proper indexes and content, as expected. So, closing, though work on improving ToC should continue.

closed 2016-06-19 by pfalcon

dpgeorge, 2016-06-28:
@pfalcon thanks for working on this!  As I said in another comment, docs.micropython.org now deletes doctrees directory before building (for both html and pdf).

pfalcon, 2016-06-29:
Thanks, but I already added "ignore cached doctree" flags for Sphinx to docs/Makefile, so everything was generated as it should last couple of weeks.