micropython/micropython #2106: micropython build fails compiling crosstool-ng

craftyguy, 2016-05-24:
When compiling micropyton following the README using the latest version from git as of about 20 minutes ago, the build fails (see output below). There was nothing in the crosstools-ng "known issues" document referenced in the output below.

Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/6.1.1/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc-multilib/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++ --enable-shared --enable-threads=posix --enable-libmpx --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --enable-multilib --disable-werror --enable-checking=release
Thread model: posix
gcc version 6.1.1 20160501 (GCC) 

```
%   make
cp -f 1000-mforce-l32.patch crosstool-NG/local-patches/gcc/4.8.5/
make -C crosstool-NG -f ../Makefile _toolchain
make[1]: Entering directory '/home/clayton/src/esp-open-sdk/crosstool-NG'
./ct-ng xtensa-lx106-elf
make[2]: Entering directory '/home/clayton/src/esp-open-sdk/crosstool-NG'
  IN    config.gen/arch.in
  IN    config.gen/kernel.in
  IN    config.gen/cc.in
  IN    config.gen/binutils.in
  IN    config.gen/libc.in
  CONF  config/config.in
#
# configuration written to .config
#

***********************************************************

Initially reported by: Max Filippov <jcmvbkbc@gmail.com>
URL: http://www.esp8266.com/viewtopic.php?f=9&t=224

***********************************************************

WARNING! This sample may enable experimental features.
         Please be sure to review the configuration prior
         to building and using your toolchain!
Now, you have been warned!

***********************************************************

Now configured for "xtensa-lx106-elf"
make[2]: Leaving directory '/home/clayton/src/esp-open-sdk/crosstool-NG'
sed -r -i.org s%CT_PREFIX_DIR=.*%CT_PREFIX_DIR="/home/clayton/src/esp-open-sdk/xtensa-lx106-elf"% .config
sed -r -i s%CT_INSTALL_DIR_RO=y%"#"CT_INSTALL_DIR_RO=y% .config
cat ../crosstool-config-overrides >> .config
./ct-ng build
make[2]: Entering directory '/home/clayton/src/esp-open-sdk/crosstool-NG'
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20160523.221355
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[INFO ]  Retrieving needed toolchain components' tarballs: done in 0.06s (at 00:02)
[INFO ]  =================================================================
[INFO ]  Extracting and patching toolchain components
[INFO ]  Extracting and patching toolchain components: done in 0.31s (at 00:02)
[INFO ]  =================================================================
[INFO ]  Installing ncurses for build
[INFO ]  Installing ncurses for build: done in 48.93s (at 00:51)
[INFO ]  =================================================================
[INFO ]  Installing GMP for host
[INFO ]  Installing GMP for host: done in 56.86s (at 01:48)
[INFO ]  =================================================================
[INFO ]  Installing MPFR for host
[INFO ]  Installing MPFR for host: done in 44.13s (at 02:32)
[INFO ]  =================================================================
[INFO ]  Installing ISL for host
[INFO ]  Installing ISL for host: done in 44.69s (at 03:17)
[INFO ]  =================================================================
[INFO ]  Installing CLooG for host
[INFO ]  Installing CLooG for host: done in 9.08s (at 03:26)
[INFO ]  =================================================================
[INFO ]  Installing MPC for host
[INFO ]  Installing MPC for host: done in 19.02s (at 03:45)
[INFO ]  =================================================================
[INFO ]  Installing expat for host
[INFO ]  Installing expat for host: done in 9.76s (at 03:55)
[INFO ]  =================================================================
[INFO ]  Installing binutils for host
[INFO ]  Installing binutils for host: done in 93.88s (at 05:29)
[INFO ]  =================================================================
[INFO ]  Installing C library headers & start files
[INFO ]  Installing C library headers & start files: done in 0.04s (at 05:29)
[INFO ]  =================================================================
[INFO ]  Installing pass-2 core C gcc compiler
[INFO ]  Installing pass-2 core C gcc compiler: done in 392.78s (at 12:01)
[INFO ]  =================================================================
[INFO ]  Installing C library
[INFO ]  Installing C library: done in 43.06s (at 12:45)
[INFO ]  =================================================================
[INFO ]  Installing final gcc compiler
[ERROR]    cfns.gperf:101:1: error: 'const char* libc_name_p(const char*, unsigned int)' redeclared inline with 'gnu_inline' attribute
[ERROR]    make[4]: *** [cp/except.o] Error 1
[ERROR]    make[3]: *** [all-gcc] Error 2
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing final gcc compiler'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_gcc_core_backend[scripts/build/cc/100-gcc.sh@545]
[ERROR]  >>        called from: do_gcc_for_host[scripts/build/cc/100-gcc.sh@671]
[ERROR]  >>        called from: do_cc_for_host[scripts/build/cc.sh@56]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-53-g46f160f/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 18:19.70)
[18:20] / ct-ng:152: recipe for target 'build' failed
make[2]: *** [build] Error 2
make[2]: Leaving directory '/home/clayton/src/esp-open-sdk/crosstool-NG'
../Makefile:116: recipe for target '_toolchain' failed
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory '/home/clayton/src/esp-open-sdk/crosstool-NG'
Makefile:112: recipe for target '/home/clayton/src/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc' failed
make: *** [/home/clayton/src/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2```

dpgeorge, 2016-05-24:
It's because you are using gcc 6.1.  The bug has been fixed in crosstool-NG, see https://github.com/jcmvbkbc/crosstool-NG/issues/24, but it is pending integration into esp-open-sdk, see https://github.com/pfalcon/esp-open-sdk/pull/164

dpgeorge, 2016-05-26:
This is now resolved.  Please pull esp-open-sdk and rebuild.

closed 2016-05-26 by dpgeorge