micropython/micropython #2111: RFC enhancement to const()

peterhinch, 2016-05-26:
In device drivers it's common practice to have a substantial number of global constants e.g.:
```python
DATA_REG = const(0x23)
COMMAND_REG = const(0X24)
```
Each line causes four bytes of RAM to be used, even if the module is implemented as frozen bytecode. This is because the language requires the variables to be available to other modules even if this is not the designer's intention. Given the use of MicroPython on resource constrained platforms this seems wasteful.

Would it be feasible to enforce module locality on globals having names starting with an underscore and declared as constant? This would turn a Python naming convention into physical reality, but only when using the const() keyword.
```python
_DATA_REG = const(0x23)
_COMMAND_REG = const(0X24)
```
This would preserve the ability to make MicroPython code compatible with cPython with
```python
const = lambda x : x
```

pfalcon, 2016-05-26:
I'm -0.5 on adding any magic syntax not grounded in Python syntax.

dpgeorge, 2016-05-26:
Python does place some significance on a leading underscore for global variables: such identifiers are *not* imported when doing `from module import *`.

Could provide another name like `private_const` that implements this feature.  Or somehow make the compiler generate a constant/frozen dict that can then be put in ROM, but that's not easy.

pfalcon, 2016-05-26:
> Python does place some significance on a leading underscore for global variables: such identifiers are not imported when doing from module import *.

Ok, formulated like that, it make sense to use that syntax to adjust semantics of const() - defined vars, which is an extension anyway.

dpgeorge, 2016-05-26:
> Ok, formulated like that, it make sense to use that syntax to adjust semantics of const() - defined vars, which is an extension anyway.

+1.  I think it is the cleanest way to provide this important optimisation (I'm always wary of adding too many constants because of RAM usage).

dpgeorge, 2016-06-06:
Implemented in b1533c43669c037d04c87d1756355cdc8edd5e0a.

closed 2016-06-06 by dpgeorge