micropython/micropython #2114: struct packing zero length throws exception

ryannathans, 2016-05-27:
Python3.5
```
>>> import struct
>>> struct.pack("0s", bytearray(0))
b''
```

Micropython
```
>>> import struct
>>> struct.pack("0s", bytearray(0))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: unsupported format
```

ryannathans, 2016-05-27:
This used to work in previous versions, I'm not yet sure which commit broke it.

```
MicroPython v1.6-246-gb894551-dirty on 2016-03-23; HydraBus1.0 with STM32F4
>>> import struct
>>> struct.pack("0s", b"")
b''
>>> struct.pack("0s", bytearray(0))
b''
```

ryannathans, 2016-05-27:
This bug was introduced by commit e53fb1bf031a53ad23a45846addf14e25cebb97d

dpgeorge, 2016-05-28:
Should be fixed by 715ee9d925871f2c44632442ff42cb58c99ee9c7.

closed 2016-05-28 by dpgeorge