micropython/micropython #2115: esp8266: make DEBUG=1 fails with inline assembly error

MarkR42, 2016-05-27:
Hi,

I can build it and run it fine without DEBUG=1.

make DEBUG=1 gives this error:

```
xtensa-lx106-elf-gcc -I../lib/netutils -I. -I.. -I../stmhal -I../lib/mp-readline -I../lib/netutils -I../lib/timeutils -Ibuild -I/home/mark/unpack/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include -Wall -Wpointer-arith -Werror -ansi -std=gnu99 -nostdlib -DUART_OS=0 -fsingle-precision-constant -Wdouble-promotion -D__ets__ -DICACHE_FLASH -fno-inline-functions -Wl,-EL -mlongcalls -mtext-section-literals -mforce-l32 -DLWIP_OPEN_SRC  -O0  -g -c -MD -o build/modmachine.o modmachine.c
In file included from modmachine.c:37:0:
xtirq.h: In function 'raise_irq_pri':
xtirq.h:43:5: error: asm operand 1 probably doesn't match constraints [-Werror]
     __asm__ volatile ("rsil %0, %1" : "=a" (old_ps) : "I" (intlevel));
     ^
cc1: all warnings being treated as errors

```

I am using the SDK from https://github.com/pfalcon/esp-open-sdk.git

dpgeorge, 2016-05-27:
That function needs to be compiled with optimisation enabled.  The only way to fix it is to provide 16 versions of the function, one for each value of the "intlevel" argument.  At the moment only 1 of those functions is used so it would be ok to do that.

dpgeorge, 2016-05-29:
The esp8266 port anyway won't compile with DEBUG=1 because it doesn't fit:
```
xtensa-lx106-elf-ld: region `iram1_0_seg' overflowed by 7836 bytes
xtensa-lx106-elf-ld: region `irom0_0_seg' overflowed by 61536 bytes
```

dpgeorge, 2016-06-06:
If there is a need for DEBUG=1 on esp8266 port then we first need to move stuff from iRAM (not easy).  If that gets done then this particular issue can be opened again.

closed 2016-06-06 by dpgeorge