micropython/micropython #2116: esp8266: wifi scan and ^C Keyboard Interrupt causes reboot

MarkR42, 2016-05-27:
Steps to reproduce:

```
sta_if = network.WLAN(network.STA_IF)                  
sta_if.scan()

```
Then press ctrl-C before scan is finished.

Expected result:

REPL command line continues working normally

Actual result:

after a few seconds, a crash & reboot:

```
>>> sta_if.scan()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
KeyboardInterrupt: 
>>> 
 ets Jan  8 2013,rst cause:2, boot mode:(3,3)

load 0x40100000, len 31364, room 16 
tail 4
chksum 0x8e
load 0x3ffe8000, len 1040, room 4 
tail 12
chksum 0x8a
ho 0 tail 12 room 4
load 0x3ffe8410, len 804, room 12 
tail 8
chksum 0xb5
csum 0xb5

```

Followed by garbage, then resets into the normal command-line.

pfalcon, 2016-05-28:
This is the same underlying issues as in #2125.

dpgeorge, 2016-05-28:
I would say we should support breaking out of the scan() function using ctrl-C.  If we don't want to do this then it's a simple fix, the ETS_POLL_WHILE line becomes:
```
while (esp_scan_list != NULL) { ets_loop_iter(); }
```

pfalcon, 2016-05-28:
> I would say we should support breaking out of the scan() function using ctrl-C.

I'm not sure if we can in general case - we're being called back by OS, what can we do? Unless you mean carefully clean up everything so we can still return, and OS will still safely call callback (which will do nothing). But there're questions like what happens in scan() is called immediately again, etc.

dpgeorge, 2016-05-28:
> Unless you mean carefully clean up everything so we can still return,

Yes, it should be possible.  After all, we are the ones calling the OS (who calls us back) in the ETS_POLL_WHILE bit.  So we effectively have complete controll.

pfalcon, 2016-05-28:
No, we asked it for calling us back, and after that it will call us back as it likes, and we can't do much, except wait. Anyway, if you want to try a particular solution, please feel free to.

dpgeorge, 2016-05-29:
Should be fixed by 6707fc94ae1b3b4227c514109b31c2e2eaba67e8, which allows ctrl-C to interrupt a running scan().

closed 2016-05-29 by dpgeorge

MarkR42, 2016-05-29:
Very good work, this fix definitely seems to work for me.

I also like the increased robustness against MemoryError in scan(). I have found that some locations have a huge number of accesspoints which take too much memory to scan.

As an option, it would be really good to have partial scan results in such situations instead of MemoryError (although MemoryError is clearly preferable to a crash)