micropython/micropython #2118: "async" is pseudo-keyword in CPython, hard keyword in uPy

pfalcon, 2016-05-27:
~~~
Python 3.5.0b3 (default, Jul 14 2015, 02:26:39) 
[GCC 4.8.4] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> def async():
...     pass
... 
>>> 
~~~

vs

~~~
MicroPython v1.8-140-g2ec0ee0-dirty on 2016-05-28; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> def async():
...     pass
...     
...     
... 
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
~~~

asyncio had an important function named async() and the reason this report is being filed. However, checking the current CPython docs, it's deprecated now: https://docs.python.org/3/library/asyncio-task.html#asyncio.async , which is a great relief.

@dpgeorge , @pohmelie : FYI. Apparently, this can be closed on read.

dpgeorge, 2016-05-28:
Yes I was aware of this.  I didn't want to complicate the lexer to allow async/await to be non-keywords.  As PEP 492 states: "async and await names will be softly deprecated in CPython 3.5 and 3.6. In 3.7 we will transform them to proper keywords. Making async and await proper keywords before 3.7 might make it harder for people to port their code to Python 3."

pfalcon, 2016-05-28:
Ok, good. It's resolved in uasyncio side, so let's close this.

closed 2016-05-28 by pfalcon