micropython/micropython #2119: cd unix; make fails - missing axtls

hauptmech, 2016-05-28:
I needed to add the following to build a fresh clone

```
git submodule update --init -- lib/axtls
cd unix
make axtls
make
```

(libffi is already on my system, so `make deplibs` and initializing all the submodules is overkill)

peterhinch, 2016-05-28:
I hit this too.README.md could do with updating

pfalcon, 2016-05-28:
> I needed

Yes, just as README describes.

pfalcon, 2016-05-28:
Both README and Makefile had changes to re-optimize for "not reading instructions in full" situation. More changes may follow, though it's hard to optimize it beyond what it is - remove something, and important information will be missing, add something - and even less people will read it in full, move something - and more people will miss the important bits than on average now.

closed 2016-05-28 by pfalcon

hauptmech, 2016-05-28:
Seems good