micropython/micropython #2124: esp8266: Suspected regression after ets_poll_iter() changes per #2090

pfalcon, 2016-05-28:
See https://github.com/micropython/micropython/pull/2090#issuecomment-222325982

dpgeorge, 2016-05-28:
A quick fix might be to change the query_irq() line to be `query_irq() > 2`.  That should get back the old behaviour (since intr_lock changed intlevel to 2) but still allow to completely disable it since disable_irq sets intlevel to 15.

pfalcon, 2016-05-28:
If you're sure that'll 100% fix it, sounds good. I for one can't see that immediately, while "separate matters" approach guaranteedly decouples it. Mind that it all may work in a non-obvious way, like what you pointed regarding ets_post calling ets_unlock_intr, etc.

dpgeorge, 2016-05-29:
I tested downloading (esp->PC) a 256k file repeatedly.  I get pretty much the same results with and without the query_irq() check in ets_loop_iter().  Takes about 12-13 seconds to transfer (about 20k/s) in the best case which is around 80% of the time.  Other times it can take up to 60s (both with and without the query_irq() check).

pfalcon, 2016-05-29:
@dpgeorge : But you paid attention to what the problem symptom, right? It's port_diag.main() hanging in the middle of its output (I provided background details to for when it ~reproducibly happens for me).

dpgeorge, 2016-05-29:
It doesn't stall for me so I don't have a chance to run port_diag.main().  And you can't type into the REPL while a file transfer is ongoing, so what were your exact steps?  port_diag.main() runs successfully after the transfer finishes.

pfalcon, 2016-05-29:
> And you can't type into the REPL while a file transfer is ongoing, so what were your exact steps?

How is that? File transfer operations are in background of course.

dpgeorge, 2016-05-30:
> How is that? File transfer operations are in background of course.

Here are my steps:
```
<build and deploy latest firmware to esp>
<connect to esp via UART REPL>
<esp network is up, via router>
>>> import webrepl
>>> webrepl.start(password='')
WebREPL daemon started on ws://192.168.4.1:8266
WebREPL daemon started on ws://192.168.0.7:8266
Started webrepl in normal mode
>>> 
<I can now type commands in UART REPL just fine>
<use webrepl_cli.py to download a file>
<while it's waiting for a password I can still type Python commands in UART REPL>
<after entering password the file transfer starts, esp to PC>
<while file transfer is ongoing I can't type into UART REPL>
<once file transfer is complete any characters input into UART REPL are now processed>
```
How did you manage to do it?  Were you uploading (PC to esp)?

pfalcon, 2016-05-30:
My steps:

~~~
WebREPL daemon started on ws://192.168.4.1:8266
WebREPL daemon started on ws://192.168.1.35:8266
Started webrepl in normal mode
>>> 
WebREPL connection from: ('192.168.1.36', 39101)

>>> 
>>> 1+2
3
>>> 4+5
9
>>> 1+2
3
>>> 2+2
4
>>> 
~~~

That connection is from:
~~~
$ ./webrepl_cli.py term.js 192.168.1.35:/
put 192.168.1.35 8266
term.js -> /term.js
Password: 
Sent 22528 of 157214 bytes
~~~

Immediately after start of transfer, and as it slowly having progress, I was able to type what you see above. 22528 is where it is still stuck as I write this. Now I wonder in what state underlying PCB is, and knowing that port_diag dumps it for me, I leisurely do:

~~~
import port_diag
~~~

I notice hiccup it the output, but maybe for 0.3s. I repeat `port_diag.main()` dozen of times, and see regular hiccups, but not longish hangs as I experienced before. In the meantime, transfer progressed to 23552. And now I get hiccup for about a second. Finally, next time I get a lockup as I used to:

~~~
Networking:
STA ifconfig: ('192.168.1.35', '255.255.255.0', '192.168.1.1', '195.175.39.40')
AP ifconfig: ('192.168.4.1', '255.255.255.0', '192.168.4.1', '195.175.39.40')
Free WiFi driver buffers of type:
0: 8
~~~

Cursor is at "0" of last line, i.e. it send "\r", but stuck before sending "\n".

UART output is stuck for 10 mins now, while webrepl xfer had an unexpected progress:

~~~
Sent 27648 of 157214 bytes
~~~

to be continued...

pfalcon, 2016-05-30:
Eventually (see comment timestamps for actual durations):
~~~
Sent 31744 of 157214 bytes
Traceback (most recent call last):
  File "./webrepl_cli.py", line 225, in <module>
    main()
  File "./webrepl_cli.py", line 219, in main
    put_file(ws, src_file, dst_file)
  File "./webrepl_cli.py", line 120, in put_file
    ws.write(buf)
  File "./webrepl_cli.py", line 44, in write
    self.s.send(hdr)
socket.error: [Errno 110] Connection timed out
~~~

UART REPL still hangs.

pfalcon, 2016-05-30:
With 
~~~
     if (query_irq() != 0) {
         return false;
     }
~~~
commented, I can't reproduce hangs on port_diag.main() - there're hiccups with output, yes, but not lockups. And file transfer is more viable - I've so far run 1 to completion, and the next one is:
~~~
Sent 98304 of 157214 bytes
~~~

Hence, the recommendation as in https://github.com/micropython/micropython/issues/2131

pfalcon, 2016-05-30:
Ah, and right, esp -> PC transfers do block the UART REPL, they're coded that way, we discussed various implications of how get vs put is currently coded, and nobody thought of such implication of current get algorithm ;-). It should be redesigned.

dpgeorge, 2016-06-01:
I tested this as best I could but I see no difference in my tests if the query_irq() if-block is commented or not.

I repeatedly transferred a 256k file to the esp with webrepl_cli.py.  All times it transferred successfully with no noticeable delay/pause (takes about 30s).  While transferring I was able to run commands at the UART REPL (eg arithmetic) and import port_diag and run port_diag.main() a few times.  Everything worked as expected, except that sometimes port_diag.main() would pause its output and wait until the file transfer was complete.  After the file transfer was done it would print everything successfully and the UART REPL would be back again.  Sometimes though the webrepl_cli.py would not exit once the transfer was complete.  Regardless, I could always transfer another file without resetting the esp8266.

The same behaviour was observed with and without the query_irq() if-block.

pfalcon, 2016-06-01:
I'm from my side, performed detailed testing above and saw correlation between the change to ets_poll_iter() and lockup during large UART output, and could not reproduce it, even with a hundred of repetitions (it was enough a dozen to trigger the issue with changes ets_poll_iter). As we're looking towards a release, and shipped previously release with funny issues, the best recommendation I have is for us to stay on safe side and redo problematic change in a way which only related to disabling interrupts from MicroPython side, not to any disabling of interrupts.

dpgeorge, 2016-06-01:
Please see 927388e80ecac345147597c1d7a53b0c1848f609.  It works for me but please test on your side.

pfalcon, 2016-06-01:
Ok, today there must be different WiFi propagation pattern (as I hinted above, less than ideal signal level was [likely] part of the test setup above), and I see behavior exactly as you describe. First of all, 150K transfer actually finish in reasonable time. Then, I got couple of times blocking of port_diag.main(), but until xfer was finished. I also saw webrepl_cli.py hanging waiting for last operation complete ack after successful content transfer.

Well, I guess everything what could be done to minimize additional issues on users' side was done, and this can be closed.

closed 2016-06-01 by pfalcon